KLayout/klayout #1748: Made DXF file on autoCAD - will not open on KLayout

cw162, 2024-06-19:
Hello, 
I have a drawing I made in AutoCAD for a photomask in microns, attached here as drawing2 in the zipfile. 
[Drawing2.zip](https://github.com/user-attachments/files/15905354/Drawing2.zip)

It looks like this 
[image]

I need to open it in KLayout as that is the software in the cleanroom. However, I kept getting the "Coordinate value overflow" error. 

I've tried to play around with the file/reader options for opening the DXF and the only database unit that works is setting to 1 um
[image]
but once it does get it open, it just looks like a giant purple circle and not my pattern, 

[image]

Can anyone please help? I just want to open my file on Klayout

Kazzz-S, 2024-06-20:
### 1. Introduction ###

**My Work Environment**
1. OS: Linux Mint 20.3
2. KLayout: 0.29.2 (source build)
3. CAD for DXF: QCAD Professional 3.30.1

**Observations**
1. At a glance, the DXF file size (300MB) looks too huge compared with the complexity of the shapes drawn.
2. I could reproduce the problems described above.
3. The diameter of the giant purple circle is `140 meters`!
4. I copied and saved a part of the design (ex. `AREA 00`) to another DXF, whose size is only 253KB.
5. Therefore, the DXF file must contain plenty of invisible (and harmful for KLayout) data.

To be continued.

Kazzz-S, 2024-06-20:
Cont.

### 2. Generating Leaner DXFs ###

* I selected ALL items and copied them to an empty sheet to generate `Drawing2A.dxf`.
* The file size is reduced to 1/100 of the original size.
* I manually edited (described later) `Drawing2A.dxf` to generate `Drawing2B.dxf`.
[image]

To be continued.

Kazzz-S, 2024-06-20:
Cont.

### 3. Drawing2A.dxf and Option-[1|2] ###
[image]
* With above Option-1, DXF ---> OASIS conversion was successful.
* However, one rectangle is missing, as shown below.
[image]

----

[image]
* With above Option-2, DXF ---> OASIS conversion was also successful.
* The rectangle appears, but is it OK?
[image]
[image]

* The rectangle consists of `1 closed poly-line` and `4 individual lines`.
* The latter are connected to a single closed poly-line while reading.
* Then, when converting to polygons (with inside-outside discrimination), 2 overlapping poly-lines cancel each other and disappear.

To be continued.

Kazzz-S, 2024-06-20:
Cont.

### 4. Drawing2B.dxf and Option-1 ###
[image]
* I deleted the `4 individual lines` to generate `Drawing2B.dxf`.
* With above Option-1, DXF ---> OASIS conversion was successful.
* The output is now OK, as shown below.
[image]

Kazzz-S, 2024-06-20:
Cont.

### 5. Summary ###
1. I suggest first cleaning up the original design in DXF.
2. Switching the reader options (as seen above) would help to detect potential errors.
3. All relevant files ===>  [GitHub1748.zip](https://github.com/user-attachments/files/15908793/GitHub1748.zip)

Regards,
Kazzz-S

cw162, 2024-06-20:
Thank you very much for looking into the problem and troubleshooting! The problem was a lot more complex that I would have thought. Ill likely stick to Klayout for now as you suggest

closed 2024-06-20 by cw162