KLayout/klayout #1749: RecursiveInstanceIterator cell_index

frankxu-omni, 2024-06-20:
I have nested.oas looks like this.
[image]

code:
[image]

ouput:
[image]
[iter_cell_index.zip](https://github.com/user-attachments/files/15917893/iter_cell_index.zip)

Kazzz-S, 2024-06-21:
My understanding is as follows.
I've slightly modified the code.

[image]
[image]
[image]

* PPT ===> [gh1749-KazzzS.zip](https://github.com/user-attachments/files/15921870/gh1749-KazzzS.zip)

frankxu-omni, 2024-06-21:
Hi Kazunari,
Your explanation is clear.  I see now. I was confused by "current instance sits in".  That really means its parent cell. 
Good graphical demo. I like that. 
Thanks, 
-Frank

closed 2024-06-21 by frankxu-omni