KLayout/klayout #1754: CellIinstArray with dtrans

frankxu-omni, 2024-06-23:
I have a nested layout: TOP->CELL_4->CELL_3->CELL_2->CELL_1. Each cell has a y-offset placement equal to 1 from CELL_1 to CELL_3. CELL_4 is placed at the origin of TOP. Every cell is R0. I want to populate the CELL_1 placement to TOP. I can see that the displacement value is 3.0, which is expected. However, when I load the OAS output into KLayout, the displacement is 0.003. Am I missing something here?
[image]
[image]
[image]
[cellinstarr.zip](https://github.com/user-attachments/files/15945607/cellinstarr.zip)

sebastian-goeldi, 2024-06-23:
Your `db.CellInstArray` should be a `db.DCellInstArray`. The official API doesn't support `CellInstArray` with `DTrans`. But since it's handled by C++ in the background, there will be an implicit cast of `DTrans` -> `Trans`. These implicit casts of transformations use `dbu=1` (i.e. `1dbu==1um`), that's why you only get a displacement by 3nm instead of 3um , because `3um`->(dbu=1)->`3dbu`->(dbu=0.001 in your layout, which is default for klayout)->`3nm`

frankxu-omni, 2024-06-24:
Hi Sebastain, Thanks for clarification.

closed 2024-06-24 by frankxu-omni