KLayout/klayout #1755: Adding intralayer connectivity crashes Klayout in large files

aq-lab, 2024-06-24:
I am working with two simultaneous .gds layouts: one is small with few layers, one is complex with many layers and devices. I have a custom device class, and whenever I enable either of the commented lines

> def get_connectivity(layout, layout_layers)    
>     conn = RBA::Connectivity::new
>     conn.connect(layout_layers[@L_layer_id],layout_layers[@L_layer_id])
>     # conn.connect(layout_layers[@C_layer_id],layout_layers[@C_layer_id])
>     # conn.connect(layout_layers[@C_layer_id])
>     conn.connect(layout_layers[@L_layer_id],layout_layers[@C_layer_id]) 
>   end

, I get the following error message in only the larger file, not the smaller one (even though the device in question is on the same layers in each):

```
ERROR: Signal number: 11
Address: 0x4e
Program Version: KLayout 0.29.2 (2024-06-10 r0436d0ccd)

Backtrace:
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x20e90a _ZN3lay25enable_signal_handler_guiEb
/usr/lib/system/libsystem_platform.dylib +0x3fdd _sigtramp
/Applications/klayout.app/Contents/Frameworks/libklayout_db.0.dylib +0x1019b63 _ZNK2db12Connectivity8interactERKNS_4CellES3_
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x3bad2 _ZN3rba15RubyInterpreter19remove_exec_handlerEPN3gsi16ExecutionHandlerE
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x3233e _ZN3rba15RubyInterpreter19remove_exec_handlerEPN3gsi16ExecutionHandlerE
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0xa8b0 _ZNK3rba22RubyStackTraceProvider11stack_depthEv
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16eda4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x15ef1a rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16bc08 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x167a73 rb_yield_1
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5c8c0 rb_protect
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x2d0da _ZN3rba15RubyInterpreter19remove_exec_handlerEPN3gsi16ExecutionHandlerE
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0xa9cb _ZNK3rba22RubyStackTraceProvider11stack_depthEv
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16eda4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x15edfb rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16bc08 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x167a73 rb_yield_1
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5c8c0 rb_protect
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x2d0da _ZN3rba15RubyInterpreter19remove_exec_handlerEPN3gsi16ExecutionHandlerE
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0xa9cb _ZNK3rba22RubyStackTraceProvider11stack_depthEv
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16eda4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x15edfb rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16bc08 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x1683f9 rb_f_eval
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x168b57 rb_obj_instance_eval
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16eda4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x15ef1a rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16bc08 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x1662ec rb_vm_call0
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x174c84 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x164d12 rb_funcallv
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5c8c0 rb_protect
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x2cef4 _ZN3rba15RubyInterpreter19remove_exec_handlerEPN3gsi16ExecutionHandlerE
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x2e330 _ZN3rba15RubyInterpreter19remove_exec_handlerEPN3gsi16ExecutionHandlerE
/Applications/klayout.app/Contents/Frameworks/libklayout_gsi.0.dylib +0x4f7b7 _ZNK2tl6Recipe11descriptionEv
/Applications/klayout.app/Contents/Frameworks/libklayout_gsi.0.dylib +0x4f544 _ZNK2tl6Recipe11descriptionEv
/Applications/klayout.app/Contents/Frameworks/libklayout_tl.0.dylib +0x785d3 _ZN2tl10Executable10do_executeEv
/Applications/klayout.app/Contents/Frameworks/libklayout_lym.0.dylib +0x2fd33 _ZN3lym16MacroInterpreter13execute_macroEPKNS_5MacroE
/Applications/klayout.app/Contents/Frameworks/libklayout_lym.0.dylib +0x3ea04 _ZNK3lym5Macro3runEv
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x771cc _ZN3lay15ApplicationBase3runEv
/Applications/klayout.app/Contents/MacOS/klayout +0x10875
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1cf0f _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16eda4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x15ef1a rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16bc08 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5c062 ruby_exec_node
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5bf65 ruby_run_node
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1ceaa _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/Applications/klayout.app/Contents/MacOS/klayout +0x1033b
/Applications/klayout.app/Contents/MacOS/klayout +0xfcab
/usr/lib/dyld +0x6366 start
```

I have verified this bug occurs in both 0.29.1 and 0.29.2 on Mac Sonoma.

Kazzz-S, 2024-06-25:
I've demangled (with Homebrew's `c++filt` ) the log to make it easier to read, but I have no idea what's happening.
Can you do the same test on Linux?

```
% cat the_log.txt | /usr/local/bin/c++filt > demangled.txt
```

```
ERROR: Signal number: 11
Address: 0x4e
Program Version: KLayout 0.29.2 (2024-06-10 r0436d0ccd)

Backtrace:
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x20e90a lay::enable_signal_handler_gui(bool)
/usr/lib/system/libsystem_platform.dylib +0x3fdd _sigtramp
/Applications/klayout.app/Contents/Frameworks/libklayout_db.0.dylib +0x1019b63 db::Connectivity::interact(db::Cell const&, db::Cell const&) const
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x3bad2 rba::RubyInterpreter::remove_exec_handler(gsi::ExecutionHandler*)
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x3233e rba::RubyInterpreter::remove_exec_handler(gsi::ExecutionHandler*)
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0xa8b0 rba::RubyStackTraceProvider::stack_depth() const
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16eda4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x15ef1a rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16bc08 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x167a73 rb_yield_1
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5c8c0 rb_protect
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x2d0da rba::RubyInterpreter::remove_exec_handler(gsi::ExecutionHandler*)
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0xa9cb rba::RubyStackTraceProvider::stack_depth() const
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16eda4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x15edfb rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16bc08 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x167a73 rb_yield_1
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5c8c0 rb_protect
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x2d0da rba::RubyInterpreter::remove_exec_handler(gsi::ExecutionHandler*)
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0xa9cb rba::RubyStackTraceProvider::stack_depth() const
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16eda4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x15edfb rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16bc08 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x1683f9 rb_f_eval
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x168b57 rb_obj_instance_eval
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16eda4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x15ef1a rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16bc08 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x1662ec rb_vm_call0
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x174c84 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x164d12 rb_funcallv
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5c8c0 rb_protect
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x2cef4 rba::RubyInterpreter::remove_exec_handler(gsi::ExecutionHandler*)
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x2e330 rba::RubyInterpreter::remove_exec_handler(gsi::ExecutionHandler*)
/Applications/klayout.app/Contents/Frameworks/libklayout_gsi.0.dylib +0x4f7b7 tl::Recipe::description() const
/Applications/klayout.app/Contents/Frameworks/libklayout_gsi.0.dylib +0x4f544 tl::Recipe::description() const
/Applications/klayout.app/Contents/Frameworks/libklayout_tl.0.dylib +0x785d3 tl::Executable::do_execute()
/Applications/klayout.app/Contents/Frameworks/libklayout_lym.0.dylib +0x2fd33 lym::MacroInterpreter::execute_macro(lym::Macro const*)
/Applications/klayout.app/Contents/Frameworks/libklayout_lym.0.dylib +0x3ea04 lym::Macro::run() const
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x771cc lay::ApplicationBase::run()
/Applications/klayout.app/Contents/MacOS/klayout +0x10875
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1cf0f rba::RubyInterpreter::initialize(int&, char**, int (*)(int&, char**))
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16eda4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x15ef1a rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16bc08 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5c062 ruby_exec_node
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5bf65 ruby_run_node
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1ceaa rba::RubyInterpreter::initialize(int&, char**, int (*)(int&, char**))
/Applications/klayout.app/Contents/MacOS/klayout +0x1033b
/Applications/klayout.app/Contents/MacOS/klayout +0xfcab
/usr/lib/dyld +0x6366 start
```

aq-lab, 2024-06-25:
I am not able to do the same test on Linux, but after further debugging I determined that adding the connectivity caused the error later on when I called remove_pin on a circuit object, and I recall other instances of remove_pin causing a similar memory error. Feel free to close, but it seems the real problem is that remove_pin causes memory errors.

klayoutmatthias, 2024-06-26:
Hi @aq-lab,

it's not really possible to debug the issue without a test case. 

I don't think the problem is related to the connectivity definition as this is a standard feature. Deleting pins is not standard however. I assume the problem is connected to that. Pins are not related to the device recognition connectivity, but changing the ways shapes are recognized may change the device terminals, hence netlist topology and hence trigger some bug on "delete_pins". 

The stack trace does not make much sense. I rather think there is a general issue with symbol resolution on MacOS. I think we had this case already before.

When and why are you deleting pins? Could you share the script that deletes the pins?

Maybe you can try to identify the actual operation the problem occurs in.

Thanks,

Matthias

klayoutmatthias, 2024-07-13:
Hi @aq-lab,

any updates on this issue?

Matthias