KLayout/klayout #1770: Macro development step over/into does not work in some cases

Kazzz-S, 2024-07-04:
Hello @klayoutmatthias,

The green arrow (**=>**) stops at a breakpoint in the macro development tool.
But it would not move with the step over/into button clicks in some builds.
[image]

I have found 3 out of 17 cases, as shown in the table below. [StepOver-Problem.xlsx](https://github.com/user-attachments/files/16093089/StepOver-Problem.xlsx)
[image]

I found this problem yesterday (first in Srl. No.10), but I need help determining the possible cause.
Do you have any suggestions, like increasing the verbosity level?

Best regards,
Kazzz-S

klayoutmatthias, 2024-07-04:
Hi @Kazzz-S,

thanks for reporting this. It reproduces very well.

I had this issue when modal dialog were opened during debugging, but apparently that applies always. I should not have deleted line 3767 in https://github.com/KLayout/klayout/commit/0a453b069c58f8dbf382adfa14ae7f97caab0966 :(

There is no trace one can easily turn on. The way to debug this is in C++, setting a breakpoint in this function:

[image]

The marked loop does not exit, as the "m_in_breakpoint" flag is not reset.

Matthias

Kazzz-S, 2024-07-05:
Hi @klayoutmatthias,

Thanks for the quick fix :-)
Now `Srl. No.10` works fine. 
I'll test the other two later at home.

Kazzz-S

Kazzz-S, 2024-07-05:
Cont.

I have also confirmed that `Srl. No.1` and `No.8` work fine.

Thanks
Kazzz-S

klayoutmatthias, 2024-07-05:
Very good - thanks for letting me know and thank you for this report! :)

Best regards,

Matthias

closed 2024-07-07 by klayoutmatthias