KLayout/klayout #1771: Python exception formatting

mole99, 2024-07-04:
Hi Matthias!

I often use KLayout together with Python (which is a great combination!), but one point that could be improved are the error messages. KLayout prints the error message, but not where it comes from. 

I haven't yet been able to figure out in the codebase where the exceptions are being printed. I assume you use a `try`/`except` each time you call user code to catch all exceptions and print them as follows:

```
try:
    1/0
except Exception as e:
    print(e)
```

Is this correct?

If you use the `traceback` module, you can print the exception in the same format as the default handler does:

```
import traceback

try:
    1/0
except Exception:
    traceback.print_exc()
```

Maybe this could be used in KLayout?

Leo

klayoutmatthias, 2024-07-04:
Hi @mole99,

do you have an example?

I am basically doing the traceback thing (in C++), but exceptions that are carried through the C++ level may loose the original location for example.

Plain Python should work. Like here:

In the debugger:

[image]

As message:

[image]

This is Ubuntu 22. I hope it's not a platform thing.

Matthias

mole99, 2024-07-05:
Hi Matthias,
thanks for your reply!

My use case is when I run pymacros in KLayout without a debugger.

For a `1/0` I get the following output in the terminal:

```
ERROR: ZeroDivisionError: division by zero
   (class ZeroDivisionError)
```

There is no reference to the location or module, so it's hard to figure out where the exception occurred.

When I run this in my system Python interpreter, however, I get:

```
Traceback (most recent call last):
  File "/path/to/file.py", line 1, in <module>
    1/0
    ~^~
ZeroDivisionError: division by zero
```

Not only the exception is printed, but also its location and module.

Interestingly, different types of errors such as `IndentationError` print the file and line number in KLayout:

```
ERROR: IndentationError: expected an indented block after function definition on line 3 (SKY130.lym, line 5) (class IndentationError)
```

That's probably because it is not an exception and handled differently.

But even then it would be nice if the formatting was the same as in the standard Python interpreter :)

Leo

klayoutmatthias, 2024-07-05:
Hmm ... that has to be some special case. Maybe in some places, the trace isn't printed.

I see stack traces all over. For example on Ubuntu with the upcoming 0.29.3 version, but it is the same with 0.29.2 on Windows. I am using this Macro:

[image]

I run outside debugger, by showing this macro in menu:

[image]

In this case I can close the debugger and run the macro through "Macros/name of macro.lym". In that case, I see a popup

[image]

When I click on "Details" I get the stack trace:

[image]

On the console, it prints the stack trace too:

[image]

And for Windows users, there is the log viewer from "File/Log Viewer" and it also shows the stack trace:

[image]

I assume you refer to a specific use case - for example when coding PCells. Also, Qt slots or event handlers may behave differently. But plain Python scripts should show the stack trace.

Matthias

mole99, 2024-07-05:
Hi Matthias, yes this is exactly my use case ^^ I would like to get Taher's PCells working, but there are some issues with them.

For me it is enough to add a pymacro like so to reproduce the issue:

```xml
<?xml version="1.0" encoding="utf-8"?>
<klayout-macro>
 <description/>
 <version/>
 <category>pymacros</category>
 <prolog/>
 <epilog/>
 <doc/>
 <autorun>true</autorun>
 <autorun-early>true</autorun-early>
 <shortcut/>
 <show-in-menu>true</show-in-menu>
 <group-name/>
 <menu-path/>
 <interpreter>python</interpreter>
 <dsl-interpreter-name/>
 <text>

1/0

 </text>
</klayout-macro>
```

When KLayout is started from a terminal, I get the error but no trace.

But there is some strange behavior: When the code above is run, then the error is printed 3 times in the terminal on startup:

```
ERROR: ZeroDivisionError: division by zero
   (class ZeroDivisionError)
ERROR: ZeroDivisionError: division by zero
   (class ZeroDivisionError)
ERROR: ZeroDivisionError: division by zero
   (class ZeroDivisionError)
```

Also, the log viewer is empty. (Maybe `autorun-early` is too early for the log viewer.)

If I disable `autorun-early`, then it is only printed once in the terminal on startup:

```
ERROR: ZeroDivisionError: division by zero
   (class ZeroDivisionError)
```

And the log viewer shows the error, but no trace:

[image]

If I run the macro manually from the GUI, I get:

[image]

Again, no stack trace. Any ideas what I could try?

Btw. I am on Debian trixie and compiled master (0.29.2). The Debian package and the flatpak package also behave the same.

Leo

klayoutmatthias, 2024-07-05:
When I pull the Debian:trixie docker image and install "python3", I get 3.11.9. Is that your version too?

Thanks,

Matthias

mole99, 2024-07-05:
Hi, yes I can confirm that is my version too!

```
Python 3.11.9 (main, Apr 10 2024, 13:16:36) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
```

klayoutmatthias, 2024-07-05:
Let me try if I can reproduce the problem with the `Debian:trixie`/3.11.9 combination.

Matthias

mole99, 2024-07-05:
On Debian bookworm with `Python 3.11.2` I get the same behavior (using flatpak):

[image]

klayoutmatthias, 2024-07-05:
Reproduces immediately on Debian, not on Ubuntu :(

At least I can debug the issue now.

Thanks, 

Matthias

mole99, 2024-07-05:
That's good to hear! (At least I haven't gone mad ^^)
If I can help you in any way, please say so.
Perhaps you could point me to the place in the code where the exceptions are printed, so I can see if I can make the formatting similar to Python's stacktrace once the issue is fixed.

Thanks!

Leo

klayoutmatthias, 2024-07-05:
I found the issue. 

Since Python 3.11.7 the semantics of `PyTracebackObject` has changed slightly. There is no public API unfortunately, so integrators have to sync with Python development :(

But that is easy to fix.

Thanks for reporting the problem.

Matthias

klayoutmatthias, 2024-07-06:
Very good. 

I agree, it is easier this way :)