KLayout/klayout #1779: Feature Request: Marker Browser: make the state of the "By Cell" or "By Category" tree persistent

stefanottili, 2024-07-06:
When I open the Marker Browser, I always expand the "By Category" Tree to look at the error layers.
It would be nice if the state of the "By Cell" and "By Category" Tree's would be persistent.

klayoutmatthias, 2024-07-27:
I have a question: I think "persistent" refers to remembering the state for the next start of KLayout - i.e. not "persisted in .lyrdb files" when you save the Marker DB.

Right?

Problem is that the collapsed state of the subnodes (i.e. per-cell, per-category) is specific to the cell or category name. So it is likely not applicable to the next marker DB file you open. But I guess most important of all is the collapsed/expanded state of the top level nodes.

stefanottili, 2024-07-27:
Yes, the toplevel node state of cell and category should be persistent between different invocations. No names involved, so not specific to any error file.

closed 2024-07-29 by stefanottili