KLayout/klayout #1780: 0.29.3 - crash when deleting shapes with multiple shapes selected

klayoutmatthias, 2024-07-08:
To reproduce:

* Open a layout in editor mode, can be a flat one
* Select multiple shapes
* Use "delete"

Result: the program crashes.

closed 2024-07-08 by klayoutmatthias

Kazzz-S, 2024-07-08:
Hello @klayoutmatthias,

Please let me know if you plan to release 0.29.4 soon to address this bug.
If so, I'll skip building 0.29.3 for Mac and wait for release 0.29.4.
After some testing, I just started building, so the impact is negligible.

Regards,
Kazzz-S

klayoutmatthias, 2024-07-09:
Thanks for asking! I have actually released 0.29.4 now.

I hope you did not waste effort. I think it is okay to skip 0.29.3. 

Sorry for the trouble,

Matthias

Kazzz-S, 2024-07-09:
Thanks for the hotfix.
No worry. I'll do so.
I started building 0.29.4 at the office.

Kazzz-S