KLayout/klayout #1784: Spice throws error at midline semicolon comments

Br1tt21, 2024-07-09:
In the same fashion as #1681, some spice dialects use semicolon to accept a comment mid line, like this:
``` .param sky130_fd_pr__nfet_05v0_nvt__vth0_slope=0.0012  ; All devices ```
This currently throws error `RuntimeError: Expected a word string here: ;`
Could this be implemented in KLayout's spice reader?

klayoutmatthias, 2024-07-09:
It can, but you won't get it that cheaply :)

There are other observations about netlist features, and a code sweep deserves more than a trivial patch.

Here are some:

* CDL appears to put a slash between nets a model/subcircuit name, like here:
```
XI1 VGND VNB VPB VPWR net59 LO / sky130_fd_sc_hd__conb_1
```

* a backslash is sometimes used to escape special characters like "[":
```
X_4255_ GCDdpath0.B_reg\[106\] net136 _2156_ VGND VGND VPWR VPWR _2157_ sky130_fd_sc_hd__mux2_1
```

what else?

Matthias

Br1tt21, 2024-07-10:
Wish I could add more, but this was the only thing I ran into.  Just working with this for my internship project, so I'm not all that experienced.

On a semi-related note though, with [this](https://github.com/klayoutmatthias/sky130_klayout_pdk/blob/main/sky130_tech/tech/sky130/lvs/sky130.lvs) project I had trouble getting the LVS to run due to a coupling component ([NGSpice manual 3.3.12](https://ngspice.sourceforge.io/docs/ngspice-manual.pdf)). Ended up removing it because it didn't matter for the layout, but it's part of a model so that's not really ideal. I couldn't make an issue on that repository so I thought I'd drop it here.

Anyway if I find anything related to this point, I'll add it!

klayoutmatthias, 2024-07-10:
I see :)

No problem. Let me just wait a little and collect some more issues, if possible.

Thanks and best regards,

Matthias

klayoutmatthias, 2024-08-01:
Should not be an issue any longer with 0.29.5

closed 2024-08-01 by klayoutmatthias