KLayout/klayout #1789: Crash by add Text to hidden layer

s910324, 2024-07-12:
Hi Matthias

Recently I run in to a issue while adding text to Hidden layer.

System : Windows 11 x64 
Version tested : 0.28.5 - 0.29.4 

#### Conditionto trigger:
* 'Libraries' windows is closed
* 'Draw to Hidden layer' Tip is enabled
* Activate add text while the layer is hidden
* If activating the add text with layer shown, but switch the visibility after activating will not cause the issue

#### Crash by add text to hidden layer
https://github.com/user-attachments/assets/d7a1c92c-f224-4b7c-aa7a-a4521e7a6ff9

#### Hide layer during editing:
Hint pops twice but does not trigger crash
https://github.com/user-attachments/assets/4823ce55-c1d0-48ee-9b77-bd4f309f2e42

#### Error message
```
Exception code: 0xc0000005
Program Version: KLayout 0.29.2 (2024-06-08 r4e430ff38) AMD64

Backtrace:
0x7ffe8ea428c0 - (Qt5Core.dll) +2173120
0x7ffe8e83708d - (Qt5Core.dll) +28813
0x7ffe8e839eed - (Qt5Core.dll) +40685
0x7ffe8e83617d - (Qt5Core.dll) +24957
0x7ffe8dc9f369 - (Qt5Widgets.dll) +979817
0x7ffe8dcd92b8 - (Qt5Widgets.dll) +1217208
0x7ffe8dd056f9 - (Qt5Widgets.dll) +1398521
0x7ffe8dd0b363 - (Qt5Widgets.dll) +1422179
0x7ffe8dbd36ba - (Qt5Widgets.dll) +145082
0x7ffe8dbd4ed5 - (Qt5Widgets.dll) +151253
0x7ffe8dbf73d4 - (Qt5Widgets.dll) +291796
0x7ffec8764421 - (klayout_lay.dll) +541729
0x7ffec8764b6c - (klayout_lay.dll) +543596
0x7ffe94e892b5 - (klayout_laybasic.dll) +627381
0x7ffe94e0a472 - (klayout_laybasic.dll) +107634
0x7ffe8eb043ab - (Qt5Core.dll) +2966443
0x7ffe8dbb41eb - (Qt5Widgets.dll) +16875
0x7ffe8dca1615 - (Qt5Widgets.dll) +988693
0x7ffe8dca1745 - (Qt5Widgets.dll) +988997
0x7ffe8dd8960e - (Qt5Widgets.dll) +1938958
0x7ffe8dbf7a20 - (Qt5Widgets.dll) +293408
0x7ffe8dbb7dae - (Qt5Widgets.dll) +32174
0x7ffe8dbc00ba - (Qt5Widgets.dll) +65722
0x7ffec86f3b04 - (klayout_lay.dll) +80644
0x7ffe8ea19038 - (Qt5Core.dll) +2003000
0x7ffe8dbbe00d - (Qt5Widgets.dll) +57357
0x7ffe8dc0dd1b - (Qt5Widgets.dll) +384283
0x7ffe8dc10f37 - (Qt5Widgets.dll) +397111
0x7ffe8dbb7dae - (Qt5Widgets.dll) +32174
0x7ffec86f3b1e - (klayout_lay.dll) +80670
0x7ffe8ea19038 - (Qt5Core.dll) +2003000
0x7ffe8e1c3784 - (Qt5Gui.dll) +276356
0x7ffe8e19cd6c - (Qt5Gui.dll) +118124
0x7ffe8ea745d8 - (Qt5Core.dll) +2377176
0x7ffeb1a1f505 - (qwindows.dll) +521477
0x7ffe8ea17374 - (Qt5Core.dll) +1995636
0x7ffe8ea1fe65 - (Qt5Core.dll) +2031205
0x7ffec86f9790 - (klayout_lay.dll) +104336
0x7ff7282b1ad9 - (klayout_app.exe) +6873
0x7ffedc2914a2 - (klayout_rba.dll) +5282
0x7ffeae1d4e4e - (x64-msvcrt-ruby310.dll) +2117198
0x7ffeae1dfc6a - (x64-msvcrt-ruby310.dll) +2161770
0x7ffeae1ee364 - (x64-msvcrt-ruby310.dll) +2220900
0x7ffeae1e0687 - (x64-msvcrt-ruby310.dll) +2164359
0x7ffeae0510fd - (x64-msvcrt-ruby310.dll) +528637
0x7ffeae054eb5 - (x64-msvcrt-ruby310.dll) +544437
0x7ffedc2982d5 - (klayout_rba.dll) +33493
0x7ff7282b2619 - (klayout_app.exe) +9753
0x7ff7282b3026 - (klayout_app.exe) +12326
0x7ff7282b12ee - (klayout_app.exe) +4846
0x7ff7282b13e6 - (klayout_app.exe) +5094
0x7ffefa79257d - (KERNEL32.DLL) +75133
0x7ffefbf8af28 - (ntdll.dll) +372520
```

klayoutmatthias, 2024-07-12:
Hi @s910324,

thanks for reporting the problem.

I can see that the popup comes on very often (which is annoying), but there is no crash for me on Ubuntu.

I'll keep looking.

Matthias

s910324, 2024-07-13:
Hi Matthias, 

I've checked more platform and machine settings, and found some thing interesting.

I have successfully re-created the crash on both Windows and Linux OS on VM with clean and all default setups.

This behavior starts to occur after version 0.27.13, which I've notice that version before this, the `Object Editor options` is still a floating QDialog type pop window where after 0.27.13 this becomes a QDockWidget.

#### Old/New version Editor option window
[image]

Where this crash always occured after this `Dock` shows up, and seems to have some connection with `Dock` visibility settings that I have mensioned, especally if layer visibility changed after `Dock` have already shows up, this no longer causes crash.

This is what I have found how to trigger the issue.
* if Editor Dock is docked 
* if Editor Dock is docked `ALONE`
* if 'Draw to Hidden layer' Tip is enabled
* if Activate add text while the layer is hidden

#### An example of Docked setup
[image]

Dock setting (A) can repeatly triggers the crash, thats why previously i though was related to by `Libraries` dock visibility setting, due to my original setting is having them Docked together, so after removing one of them it caused the issue.

klayoutmatthias, 2024-07-13:
You're fantastic! :)

I managed to reproduce the problem with the editor options docked at the top:

[image]

Now debugging can start :)

Thanks,

Matthias

closed 2024-07-21 by klayoutmatthias