KLayout/klayout #1793: inconsistent results in width_check when calling by keyword argument (python)

tvt173, 2024-07-19:
Hi @klayoutmatthias ,

I found a troublesome bug today when using a keyword argument to set the `min_projection` in a `width_check` for a `Region` object in python. I find that I get consistent results when calling using a positional argument but inconsistent results when using a keyword argument. Do you know why this might be?

Please see the example script below:

```python
import klayout.db as kdb
import numpy as np

def always(value_dbu):
    # min_projection is not set
    return region.width_check(value_dbu)

def never(value_dbu):
    # setting min_projection via a positional argument
    return region.width_check(value_dbu, False, kdb.Metrics.Euclidian, np.deg2rad(90), value_dbu)

def sometimes(value_dbu):
    # setting min_projection via a named argument
    return region.width_check(value_dbu, min_projection=value_dbu)

filename = r"troublesome_taper.oas"
l = kdb.Layout()
value_dbu = 1000
l.read(filename)
layer = l.layer(8, 0)
region = kdb.Region(l.begin_shapes(l.top_cell(), layer=layer))
for func in [always, never, sometimes]:
    results = []
    for i in range(30):
        result = len(func(value_dbu))
        results.append(str(result))
    print(f"{func.__name__}: " + "".join(results))
```

Expected Output:
```console
always: 111111111111111111111111111111
never: 000000000000000000000000000000
sometimes: 000000000000000000000000000000
```

Actual Output:
```console
always: 111111111111111111111111111111
never: 000000000000000000000000000000
sometimes: 011000100001001100000100000000
```

I'm using the standalone klayout python package v0.29.2 on Windows 11. 

Sample gds: [troublesome_taper.zip](https://github.com/user-attachments/files/16302459/troublesome_taper.zip)

sebastian-goeldi, 2024-07-19:
Your `never` case might be flawed:

The ignore angle should be in degrees not radian:

From the docs:

```
"ignore_angle" specifies the angle limit of two edges. If two edges form an angle equal or above the given value, they will not contribute in the check. Setting this value to 90 (the default) will exclude edges with an angle of 90 degree or more from the check. Use nil for this value to select the default.
```

So for proper comparison it should be `90` or even better `None`

Does this change the output of `never`?

tvt173, 2024-07-19:
nice catch @sebastian-goeldi ! indeed it changes the results for the "never" case, but it does not change the fact that "sometimes" behaves inconsistently

updated script:
```python
import klayout.db as kdb
import numpy as np

def always(value_dbu):
    # min_projection is not set
    return region.width_check(value_dbu, False, kdb.Metrics.Euclidian, 90)

def never(value_dbu):
    # setting min_projection via a positional argument
    return region.width_check(value_dbu, False, kdb.Metrics.Euclidian, 90, value_dbu)

def sometimes(value_dbu):
    # setting min_projection via a named argument
    return region.width_check(value_dbu, min_projection=value_dbu)

filename = r"tests\gds\troublesome_taper.oas"
l = kdb.Layout()
value_dbu = 1000
l.read(filename)
layer = l.layer(8, 0)
region = kdb.Region(l.begin_shapes(l.top_cell(), layer=layer))
for func in [always, never, sometimes]:
    results = []
    for i in range(30):
        result = len(func(value_dbu))
        results.append(str(result))
    print(f"{func.__name__}: " + "".join(results))
```

Updated results:
```console
always: 111111111111111111111111111111
never: 111111111111111111111111111111
sometimes: 011000000000000000000000000000
```

sebastian-goeldi, 2024-07-19:
hmm, time to upgrade to 0.29.4 (or linux :P); with 0.29.4 I get:

unfixed ignore angle:
```
python test.py 
always: 111111111111111111111111111111
never: 000000000000000000000000000000
sometimes: 111111111111111111111111111111
```

fixed to `None`:
```
python test.py 
always: 111111111111111111111111111111
never: 111111111111111111111111111111
sometimes: 111111111111111111111111111111
```

tvt173, 2024-07-19:
i still get the inconsistent behavior on 0.29.4/windows 😞

sebastian-goeldi, 2024-07-19:
I just tested it in windows 11 VM vs debian, the results are interesting...

[image]

klayoutmatthias, 2024-07-19:
I am getting an even more interesting one on Ubuntu 22.04:

[image]

I think that is more or less the same problem.

Time for debugging :(

Matthias

klayoutmatthias, 2024-07-19:
I think I found the issue. 

The compiler does not warn if passing a return-by-value to "const &" .. :(

It will be easy to fix. The issue happens only for keyword parameters with an enum- or object-type default value. In this case it was the Metrics type I think.

Matthias

tvt173, 2024-07-19:
Great! Thanks for the quick work getting to the bottom of it! A patch release would be appreciated once the fix is in

closed 2024-07-27 by klayoutmatthias