KLayout/klayout #1812: new python release has issues

joamatab, 2024-08-01:
RuntimeError: Cannot write layer numbers larger than 65535 to GDS2 streams, writing cell 'Unnamed_0' in Cell.write
https://github.com/gdsfactory/gdsfactory/actions/runs/10189276913/job/28186995933

and segfaults
https://github.com/gdsfactory/gdsfactory/actions/runs/10189276906/job/28186996591

@sebastian-goeldi 
@thomaslima

sebastian-goeldi, 2024-08-01:
You only have two bytes to store layer numbers and two for datatype, so this number is correct.

klayoutmatthias, 2024-08-01:
I will take a look.

joamatab, 2024-08-01:
im still getting errors on the latest klayout

Using klayout 0.29.4 it works

```python
import gdsfactory as gf 
c = gf.components.straight(length=10, layer=(1, 0))
c.write_gds(with_metadata=False)
```

However using kayout 0.29.5

```
RuntimeError                              Traceback (most recent call last)
File ~/demo/hi.py:4
      2 import gdsfactory as gf 
      3 c = gf.components.straight(length=10, layer=(1, 0))
----> 4 c.write_gds(with_metadata=False)

File ~/c/pdks/gdsfactory/gdsfactory/component.py:841, in ComponentBase.write_gds(self, gdspath, gdsdir, sa
ve_options, with_metadata, **kwargs)
    839     for k in kwargs:
    840         warnings.warn(f"{k} is deprecated", stacklevel=2)
--> 841 self.write(filename=gdspath, save_options=save_options)
    842 return pathlib.Path(gdspath)

File ~/c/kfactory/src/kfactory/kcell.py:1562, in KCell.write(self, filename, save_options, convert_externa
l_cells, set_meta_data)
   1559     kc = self.kcl[kci]
   1560     kc.insert_vinsts()
-> 1562 self._kdb_cell.write(str(filename), save_options)

RuntimeError: Cannot write layer numbers larger than 65535 to GDS2 streams, writing cell 'straight_L10_N2_
CSstrip_L1_0' in Cell.write

```

klayoutmatthias, 2024-08-01:
Hmm ...

```
import klayout
print(klayout.__version__)

import gdsfactory as gf 
c = gf.components.straight(length=10, layer=(1, 0))
c.write_gds(with_metadata=False)

print("Done.")
```

gives for me:

```
0.29.5
Done.
```

No error :(

sebastian-goeldi, 2024-08-01:
@joamatab can you try rebuilding your env? I suspect there is some issue of an older package interfering, I also tried today and didn't get am error in a similar test

joamatab, 2024-08-01:
Sebastian, i tried in two different computers with different OS, both of them running python3.11

joamatab, 2024-08-01:
You can see the tests for different OS and python versions here

https://github.com/gdsfactory/gdsfactory/actions/runs/10203362236/job/28229538713

klayoutmatthias, 2024-08-01:
I don't see that this is related to "Cannot write layer numbers larger than 65535 to GDS2 streams", but I see it is related to freetype.

I seriously think that the problem may be triggered by KLayout, but it is not the root cause.

klayoutmatthias, 2024-08-01:
Regarding the freetype lib, I can reproduce the following weird behavior:

This segfaults at exactly the same place than the GDSFactory tests:

```
import freetype

font = freetype.Face("/usr/share/fonts/truetype/freefont/FreeSansBold.ttf")

# print(f"{font._FT_Face.contents.glyph.contents.outline}")

outline = font._FT_Face.contents.glyph.contents.outline
print(freetype.FT_Outline_Get_Orientation(outline))
```

This does not (not the print statement now is not commented):

```
import freetype

font = freetype.Face("/usr/share/fonts/truetype/freefont/FreeSansBold.ttf")

print(f"{font._FT_Face.contents.glyph.contents.outline}")

outline = font._FT_Face.contents.glyph.contents.outline
print(freetype.FT_Outline_Get_Orientation(outline))
```

That is on Ubuntu 22.04 LTS with Python 3.10.12.

I think there is something wrong with the C structure bindings.

Matthias

klayoutmatthias, 2024-08-04:
I am a bit lost now. Is there something I should fix?

joamatab, 2024-08-04:
sorry, this was a strange issue, 

it seems like for some OS or python versions we get some issues between klayout and freetype

i think we can close it for now

closed 2024-08-04 by joamatab