KLayout/klayout #1816: `interacting` operation between EdgePairs and Region (or edges)

tvt173, 2024-08-01:
Hi @klayoutmatthias ,

Sometimes it is useful to filter edge pairs based on those which interact with a Region. I know I can convert the EdgePairs to a region or edges and then run `interacting` between those two objects. However, in some cases it would be nice to keep the result of the operation as EdgePairs, and I'm not sure if there is a straightforward way to do that today. The best I could think of would be to implement an `EdgePairFilter` that checks interacting between each of the edges in the pair and Region and returns True if either are True.

What do you think about implementing this operation? I think it would be useful, and it seems pretty straightforward.

Thanks,
Troy

klayoutmatthias, 2024-08-02:
Yes, I think that makes sense and it is possible. There is a certain effort, because EdgePairs are not primary objects, but that can be changed.

klayoutmatthias, 2024-08-04:
The linked PR provides a first implementation.

Use is straightforward - the following DRC functions now also act on edge pair layers:

* `pull_interacting` (with edge and polygon layers)
* `interacting` (with edge and polygon layers, with count if needed)
* `not_interacting` (with edge and polygon layers, with count if needed)
* `split_interacting` (with edge and polygon layers)
* `inside` (with polygon layers)
* `not_inside` (with polygon layers)
* `split_inside` (with polygon layers)
* `outside` (with polygon layers)
* `not_outside` (with polygon layers)
* `split_outside` (with polygon layers)

All features work in deep or flat mode.

tvt173, 2024-08-05:
this is great @klayoutmatthias ! thank you!

closed 2024-08-10 by klayoutmatthias