KLayout/klayout #1819: Discrete soft small rendering

bbstrikesagain, 2024-08-03:
Back until 0.27.9, when I enabled **oversampling 2x** or **3x** it had the effect of shrinking the overall scale of stipples and line styles.  Oversampling didn't affect the Layers palette, so the sample swatches there were rendered at full size regardless.  That inconsistency was actually quite useful, as swatches are never overlaid in the Layers palette and full size rendering there made it clearer when editing the layer appearance.  The oversampling could be disabled to render the layout using the same full size patterns, but in complex overlays the appearance was ugly and lacked stipple translucence.  I generally left oversampling enabled to resolve layouts more in a more immersive, intuitive way, e.g:

[image]

Apparently since 0.28, and certainly in 0.29 versions, **High resolution mode** was added to support high DPI displays, but since then there's no combination of options that can turn off that feature and render the tidy scaled down oversampling.  I'm using a 94 DPI 3840x1200 monitor, but none of the new klayout options can achieve anything like what I had in 0.27.  Compare the appearance in 0.29.5:

[image]

It would be really great to have an option to support low resolution monitors and restore oversampling with the shrink

klayoutmatthias, 2024-08-04:
Very good. Thanks for this ticket! 

My proposal is an option to select "small rendering" under oversampling conditions. I will try that and provide a proposal. There is an interaction with the selection of the "Default" font glyphs and I think it makes sense to not scale the fonts with the oversampling in "small rendering" mode. I tried that, and the fonts basically become unreadable, unless you compensate that by choosing "XXXLarge" font.

Matthias

klayoutmatthias, 2024-08-04:
This is my current solution: 

[image]

And this is the effect:

**No Oversampling**

[image]

**3x Oversampling, "Small Rendering" OFF**

[image]

**3x Oversampling, "Small Rendering" ON**

[image]

closed 2024-08-10 by klayoutmatthias