KLayout/klayout #1827: Klayout 0.29.5: command + select shapes cause the program to crash

dsus4wang, 2024-08-14:
Hi!
I often use command+select some shapes to do the "Invert Selection" , now I update klayout to version 0.29.5, when I do this invert selection again, If I deselect the last shape (or if I deselect without selecting any shape), the program will crash.
Here is the Log:
```
Signal number: 11
Address: 0x0
Program Version: KLayout 0.29.5 (2024-08-05 rLatestSourcePackage)

Backtrace:
/Applications/klayout0295.app/Contents/Frameworks/libklayout_lay.0.dylib +0x1dd114 _ZN3lay25enable_signal_handler_guiEb
/usr/lib/system/libsystem_platform.dylib +0x3a24 _sigtramp
/Applications/klayout0295.app/Contents/Frameworks/libklayout_edt.0.dylib +0xdba24 _ZN3edt7Service14display_statusEb
/Applications/klayout0295.app/Contents/Frameworks/libklayout_edt.0.dylib +0xde328 _ZN3edt7Service6selectERKN2db3boxIddEEN3lay8Editable13SelectionModeE
/Applications/klayout0295.app/Contents/Frameworks/libklayout_laybasic.0.dylib +0x148e60 _ZN3lay9Editables6selectERKN2db3boxIddEENS_8Editable13SelectionModeE
/Applications/klayout0295.app/Contents/Frameworks/libklayout_laybasic.0.dylib +0x1a13ac _ZN3lay16SelectionService19mouse_release_eventERKN2db5pointIdEEjb
/Applications/klayout0295.app/Contents/Frameworks/libklayout_laybasic.0.dylib +0x1a15c4 _ZThn16_N3lay16SelectionService19mouse_release_eventERKN2db5pointIdEEjb
/Applications/klayout0295.app/Contents/Frameworks/libklayout_laybasic.0.dylib +0x1f5790 _ZN3lay12ViewObjectUI24send_mouse_release_eventERKN2db5pointIdEEj
/Applications/klayout0295.app/Contents/Frameworks/libklayout_laybasic.0.dylib +0x1f80d0 _ZN3lay22BitmapViewObjectCanvas9bg_bitmapEv
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtWidgets.framework/Versions/A/QtWidgets +0x4c15c _ZN7QWidget5eventEP6QEvent
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtWidgets.framework/Versions/A/QtWidgets +0xdad0 _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtWidgets.framework/Versions/A/QtWidgets +0xfb78 _ZN12QApplication6notifyEP7QObjectP6QEvent
/Applications/klayout0295.app/Contents/Frameworks/libklayout_lay.0.dylib +0x758f4 _ZN3lay14GuiApplication9do_notifyEP7QObjectP6QEvent
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtCore.framework/Versions/A/QtCore +0x7444c _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtWidgets.framework/Versions/A/QtWidgets +0xe070 _ZN19QApplicationPrivate14sendMouseEventEP7QWidgetP11QMouseEventS1_S1_PS1_R8QPointerIS0_Ebb
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtWidgets.framework/Versions/A/QtWidgets +0x5e7ec _ZN13QWidgetWindow16handleMouseEventEP11QMouseEvent
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtWidgets.framework/Versions/A/QtWidgets +0x5da84 _ZN13QWidgetWindow5eventEP6QEvent
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtWidgets.framework/Versions/A/QtWidgets +0xdad0 _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtWidgets.framework/Versions/A/QtWidgets +0xe938 _ZN12QApplication6notifyEP7QObjectP6QEvent
/Applications/klayout0295.app/Contents/Frameworks/libklayout_lay.0.dylib +0x75900 _ZN3lay14GuiApplication9do_notifyEP7QObjectP6QEvent
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtCore.framework/Versions/A/QtCore +0x7444c _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtGui.framework/Versions/A/QtGui +0x65b88 _ZN22QGuiApplicationPrivate17processMouseEventEPN29QWindowSystemInterfacePrivate10MouseEventE
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtGui.framework/Versions/A/QtGui +0xb2b94 _ZN22QWindowSystemInterface22sendWindowSystemEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE
/opt/homebrew/Cellar/qt/6.6.2_1/share/qt/plugins/platforms/libqcocoa.dylib +0x14fb4 _ZN28QCocoaEventDispatcherPrivate26postedEventsSourceCallbackEPv
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7f5cc __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7f560 __CFRunLoopDoSource0
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7f2d0 __CFRunLoopDoSources0
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7ded8 __CFRunLoopRun
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7d448 CFRunLoopRunSpecific
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x30df0 RunCurrentEventLoopInMode
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x30a80 ReceiveNextEventCommon
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x30984 _BlockUntilNextEventMatchingListInModeWithFilter
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x3997c _DPSNextEvent
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x38b18 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x2cf7c -[NSApplication run]
/opt/homebrew/Cellar/qt/6.6.2_1/share/qt/plugins/platforms/libqcocoa.dylib +0x1340c _ZN21QCocoaEventDispatcher13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtCore.framework/Versions/A/QtCore +0x7bc00 _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE
/opt/homebrew/Cellar/qt/6.6.2_1/lib/QtCore.framework/Versions/A/QtCore +0x7494c _ZN16QCoreApplication4execEv
/Applications/klayout0295.app/Contents/Frameworks/libklayout_lay.0.dylib +0x76164 _ZN3lay14GuiApplication4execEv
/Applications/klayout0295.app/Contents/Frameworks/libklayout_lay.0.dylib +0x74bcc _ZN3lay15ApplicationBase3runEv
/Applications/klayout0295.app/Contents/MacOS/klayout +0x12f10
/Applications/klayout0295.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1b294 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/opt/homebrew/Cellar/ruby/3.3.0/lib/libruby.3.3.dylib +0x1d0224 vm_call_cfunc_with_frame_
/opt/homebrew/Cellar/ruby/3.3.0/lib/libruby.3.3.dylib +0x1ba7f0 vm_exec_core
/opt/homebrew/Cellar/ruby/3.3.0/lib/libruby.3.3.dylib +0x1b7d48 rb_vm_exec
/opt/homebrew/Cellar/ruby/3.3.0/lib/libruby.3.3.dylib +0x80f4c rb_ec_exec_node
/opt/homebrew/Cellar/ruby/3.3.0/lib/libruby.3.3.dylib +0x80e28 ruby_run_node
/Applications/klayout0295.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1b220 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/Applications/klayout0295.app/Contents/MacOS/klayout +0x12898
/Applications/klayout0295.app/Contents/MacOS/klayout +0x12044
/usr/lib/dyld +0x5f28 start
```
Could you help me to check this problem? thanks a lot!

klayoutmatthias, 2024-08-14:
Hi @dsus4wang,

I think this is related to this bug: https://github.com/KLayout/klayout/issues/1814. 
I will release a new version very soon which fixes this bug.

If you want to test that: you will find some up-to-date builds here: https://www.klayout.org/downloads/master/ (disclaimer: not signed).

Matthias

Kazzz-S, 2024-08-14:
Hello @klayoutmatthias,

I could reproduce this problem with `ST-klayout-0.29.5-macOS-Monterey-1-qt5MP-RsysPsys.dmg`, which is just after the `d1121fae` commit.

I shot the video on Linux.

https://github.com/user-attachments/assets/7588307d-cf92-4eff-b0e9-87a5bc78062f

I selected the first rectangle (Shift+click), the second rectangle (Shift+click), and the triangle (Shift+click), then deselected the triangle (Ctrl+click). The operation then froze.

I have confirmed the problem is fixed (on Mac) with the latest `9f16f071` commit (version 0.29.6-to-be).

Thanks
Kazzz-S

klayoutmatthias, 2024-08-17:
@dsus4wang As the new version is released, can I close this issue?

Thanks,

Matthias

closed 2024-08-19 by dsus4wang