KLayout/klayout #1828: feature request: "move to" function for "partial selection"

dsus4wang, 2024-08-14:
Hi @klayoutmatthias ,
The partial function is a great function! I often use it to change the position of graphic parts
But sometimes I need to change the position precisely, Is that possible to add the precise move (like "move to" for shape selection) for the partial selection? If not, Is there a way to get the partial(points and edges I think) like layout_view.each_object_selected()?
Thanks!

klayoutmatthias, 2024-08-14:
Hi @dsus4wang,

this is already there :)

Switch to Partial mode, select edges by dragging the selection box with the mouse and left button clicked. Use "Shift" + drag to add more edges. Then use "Edit / Selection / Move To". You can also use "Move By" to shift the selection by a numeric distance.

Partial selection is not available to "each_object_selected". That is a complex topic and manipulation of partial selection is beyond the scope of the scripting API.

Matthias

dsus4wang, 2024-08-16:
Haha. I tried. Yes!
Thanks!

closed 2024-08-16 by dsus4wang