KLayout/klayout #1831: why is not work

knightzhong, 2024-08-20:
Hello,

I've been using the smooth function with the keep_hv parameter set to True, and I've noticed that regardless of how large I set the d value, the function does not seem to make any changes to the polygon. I am attempting to smooth out the polygon while maintaining horizontal and vertical edges, but it appears that the function is not responding to the changes in the d parameter.

Could you please advise why this might be happening? Is there a specific reason why the smooth function is not altering the polygon even when a large d value is specified with keep_hv=True?
mycode:all_image_geo_smooth = all_image_geo_smooth.smoothed(pixel_size *100,keep_hv = True)
Thank you for your time and assistance.
[image]
[image]

sebastian-goeldi, 2024-08-20:
You say you want to preserve all horizontal or vertical edges but complain the smooth function when set to preserve horizontal and vertical edges.

I think you might misunderstand the flag. The flag doesn't day "all the edges should be horizontal or vertical". It means " do not make horizontal and vertical edges shorter or make them non-horizontal/vertical edges".

If you want to simply smooth but keep the shapes (assuming you only have pixel like structures here). You should look into the `Region.size`. It should do what I think it is what you want to do. In more advanced cases you might need `Region.minkowski_sum` but that one requires more setup and tricks.

Anyway:

`reg.size(pixel_size*100).size(-pixel_size*100)` is likely what you want for the pixel problems.

Essentially do

knightzhong, 2024-08-21:
I'm very sorry, I misunderstood the function. Yes, thank you for your answer. I want to keep both the horizontal and vertical lines smooth while smoothing my polygon, so that it doesn't have so many bumps and hollows. What can I do?

sebastian-goeldi, 2024-08-22:
Did you try the thing I suggested at the end with (assuming your region is called `r`) `r.size(100*pixel_size).size(-100*pixel_size)`?

That is very likely what you want and will work very well for manhattanized structures.

klayoutmatthias, 2024-08-22:
@sebastian-goeldi Thanks for the suggestion. The "smooth" function is based on vertex elimination and removing a vertex will render diagonal edges. As you rightly pointed out, that is not compatible with "keep_hv". This option is intended for partially-Manhattan polygons where the vertical and horizontal edges should be preserved. This is important to keep stitch lines adjacent for example. With a pure manhattan polygon and "keep_hv" as True, there is no degree of freedom for the vertex elimination.

I wonder in general, whether smoothing a pixel image in polygon space is a good idea. I think there are better methods for smoothing in pixel space. For example, you could apply a low-pass filter to the image and a threshold to render a smoother contour.

Matthias