KLayout/klayout #1834: Klayout Crash when trying to install technology libraries

Revenantx86, 2024-08-27:
Hello, I get the following crash report when I try to run sky130 technology library by adding through technology manager. I also get crash when I install gf180nm pdk using built-in technology manager.

Signal number: 6
Address: 0x3e800004e0a
Program Version: KLayout 0.28.8 (2023-05-23 rfa9e7e7c1)

Backtrace:
/usr/lib/klayout/libklayout_lay.so.0 +0x2e0827 lay::enable_signal_handler_gui(bool) [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x42520 __sigaction [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x969fc pthread_kill [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x42476 raise [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x287f3 abort [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x89676 __fsetlocking [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0xa0cfc timer_settime [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0xa0fdc timer_settime [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0xa549a free [??:?]
/home/unixpoly/.local/lib/python3.10/site-packages/klayout/lib_tl.cpython-310-x86_64-linux-gnu.so +0x60f4e __cxa_throw_bad_array_new_length [??:?]
/lib64/ld-linux-x86-64.so.2 +0x647e __nptl_change_stack_perm [??:?]
/lib64/ld-linux-x86-64.so.2 +0x6568 __nptl_change_stack_perm [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x174af5 _dl_catch_exception [??:?]
/lib64/ld-linux-x86-64.so.2 +0xdff6 _dl_rtld_di_serinfo [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x174a98 _dl_catch_exception [??:?]
/lib64/ld-linux-x86-64.so.2 +0xe34e _dl_find_dso_for_object [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x9063c dlmopen [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x174a98 _dl_catch_exception [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x174b63 _dl_catch_error [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x9012e dlerror [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x906c8 dlopen [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x228330 _Py_UTF8_Edit_Cost [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1e901d _PyImport_FixupExtensionObject [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1287b3 _PyModuleSpec_IsInitializing [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x76776 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7b9c8 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x79efe _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7863e _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7863e _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7863e _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0xe13f4 PyObject_CallNoArgs [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0xe175f _PyObject_CallMethodIdObjArgs [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1eb2a4 PyImport_ImportModuleLevelObject [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7a715 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1bd3de PyEval_EvalCode [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1bd96d PyEval_EvalCode [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1287b3 _PyModuleSpec_IsInitializing [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x76776 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7b9c8 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x79efe _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7863e _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7863e _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0xe13f4 PyObject_CallNoArgs [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0xe175f _PyObject_CallMethodIdObjArgs [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1eb2a4 PyImport_ImportModuleLevelObject [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7a715 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1bd3de PyEval_EvalCode [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1bd96d PyEval_EvalCode [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1287b3 _PyModuleSpec_IsInitializing [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x76776 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7b9c8 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x79efe _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7863e _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7863e _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0xe13f4 PyObject_CallNoArgs [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0xe175f _PyObject_CallMethodIdObjArgs [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1eb2a4 PyImport_ImportModuleLevelObject [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1be2fe PyEval_EvalCode [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x128023 PyMemoryView_GetContiguous [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0xe1fec _PyObject_Call [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x76776 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7863e _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7863e _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0xe13f4 PyObject_CallNoArgs [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0xe175f _PyObject_CallMethodIdObjArgs [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1eb2a4 PyImport_ImportModuleLevelObject [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7a715 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1bd3de PyEval_EvalCode [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1bd96d PyEval_EvalCode [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1287b3 _PyModuleSpec_IsInitializing [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x76776 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7b9c8 _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x79efe _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x7863e _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c23af _PyEval_SetAsyncGenFinalizer [??:?]

klayoutmatthias, 2024-08-27:
0.28.8 is older and not very stable. Did you try with a later version?

Also, at least 0.28.13 is required for using git-managed packages with are becoming common now.

Matthias

Revenantx86, 2024-08-28:
> 0.28.8 is older and not very stable. Did you try with a later version? My recommendation is 0.29.6 which I think is a pretty good one.
> 
> Also, at least 0.28.13 is required for using git-managed packages which are becoming common now.
> 
> Matthias

Thank you very much. It was because I was running older version. I have upgraded it to latest version and it works now.

closed 2024-08-28 by Revenantx86