KLayout/klayout #1856: KLayout 0.29.4 is not responding

olisnr, 2024-09-14:
from time to time i have an "KLayout 0.29.4 is not responding" dialog, and need to end the process. i cant reproduce it, because i didnt know what hotkey i was pressing by mistake. but now i only wanted to close the layouts, after experimenting with the scripts of iHPs openPDK. i got an error message in the terminal:

```
...
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : Starting Taps EXTRACTION
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : Extracting ntap1 device
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : Extracting ptap1 device
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : Starting SG13G2 LVS Options Preparations
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : Starting SG13G2 LVS Alignment
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : Starting SG13G2 LVS Simplification
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : Starting SG13G2 LVS Comparison
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : ERROR : Netlists don't match
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
2024-09-11 20:58:34 +0200: Memory Usage (2434364K) : LVS Total Run time 7.034608 seconds
malloc_consolidate(): unaligned fastbin chunk detected
```

is there a debug mod that logs someting You want to know, if it happens again?

klayoutmatthias, 2024-09-15:
Please try with the latest version first.

olisnr, 2024-09-16:
thanks. i download the newest version. about logging, there is nothing i should enable?

closed 2024-09-16 by olisnr

sebastian-goeldi, 2024-09-16:
> thanks. i download the newest version. about logging, there is nothing i should enable?

See here `-d` https://www.klayout.de/command_args.html

klayoutmatthias, 2024-09-16:
I don't think logging will help a lot. I just mentioned the latest version as it is worth checking if there is a bug that was fixed.

Maybe there is one thing that is worth checking: on some platforms, the application will hang in the way you noticed when you close the application while the properties dialog is open (https://github.com/KLayout/klayout/issues/1857).

Matthias