KLayout/klayout #1863: DRC: How to find a "Path" with 0 width?

MingChingCheng, 2024-09-19:
Hi developers and all,

KLayout version: 0.29.6
OS: Windows 10

I am using KLayout to read a DXF file and transfom it into GDS format.
If the polyline in DXF file is not closed, KLayout will create a "Path with 0 width" instead of a "Polygon".
[image]

I would like to find a way to check if there is any "Path with 0 width" in my layout. 

Here is an example, I drawn a Path with 0 width at the left, and a Path with 0.5 width at the right.
[image]

I have tried using "width" expression with DRC code:
[image]

The result shows as following.
[image]

You can see the Path with 0 width will not be checked. 
I have also tried to using "area" expression, but it didn't work also.
Does anyone know how should I do? 

Thanks a lot!

klayoutmatthias, 2024-09-19:
This is not a bug - DRC is not made for zero-area objects. It is made for polygons. A line is not a polygon and will simply be ignored. DRC width or area is not the right tool for finding such issues.

I'd propose to use a script that walks over all cells, layers and shapes and identifies the zero-width paths.

MingChingCheng, 2024-09-20:
Dear Matthias,

Thanks for your reply!
I will try to use script to do this work.

sebastian-goeldi, 2024-09-20:
You can use something like this (this is for python, ruby would require ruby syntax but should be very similar):

```python
import pya
ly # layout object, you can grab this from the active cellview if you want to use the GUI current application
cells = ly.cells("*") # you could also only grab specific cells
layer_indexes = ly.layer_indexes # again could restrict this and grab on specific ones with ly.layer
violating_shapes = []
for cell in cells:
    for layer inayer_indexes:
        for shape in c.shapes(layer).each(pya.Shapes.SPaths): # SPaths could also be SRegions if you want to consider all polygons, in which case you would need shape.polygon.area() == 0
            if shape.path.width == 0
                 violating_shapes.append(shape)
```