KLayout/klayout #1867: Request: wheel for linux-aarch64

stevej, 2024-09-23:
For Tiny Tapeout, we've been working on a devcontainer that contains klayout among other tools. A devcontainer is a Docker-based development environment for cross-platform and web use. The problem we're running into is that on Docker for macOS, building the devcontainer requires a full build of klayout from source for linux-aarch64 which can exceed the default memory limit that Docker for macOS ships with. Having a pre-built wheel would not only speed up building the container but also avoid this memory limitation. Thanks!

klayoutmatthias, 2024-09-25:
I have just committed https://github.com/KLayout/klayout/commit/8a5754c11258a4d421de8fcf62a4df864111d65a. To my understand that should enable ARM64 builds on Linux. Let's see where this leads to.

Keeping you updated.

Matthias

klayoutmatthias, 2024-09-25:
I guess I need a little support ...

I am not sure if GitHub actually offers ARM64 Linux runners. I can't see how I can enable that. 

Is there a chance to get such a runner from GitHub, or are self-hosted runners the only available option? I don't have ARM64 hardware and an emulator rules out I guess.

Any advice?

Matthias

klayoutmatthias, 2024-09-28:
Hi @stevej,

Thanks for this information. I'd say we leave this ticket open. As soon as ARM64 runners are available, it will be easy to enable them.

Matthias

adrienluitot, 2024-11-30:
GitHub runners for Linux Arm64 should arrive soon indeed. But if you want to try before GH release them, you can use CircleCI. They offer Arm64 runners for free, I tried it for another project, it works well! You can have a look at the config I used: https://github.com/adrienluitot/OpenVAF/blob/adb199a731e3cd5b9f14497e5f0253042e67b398/.circleci/config.yml

klayoutmatthias, 2024-12-01:
GitHub runners are only good for building the Python modules. Building the application needs too much resources.

stevej, 2025-08-12:
Github arm64 runners are finally in GA. https://github.blog/changelog/2025-08-07-arm64-hosted-runners-for-public-repositories-are-now-generally-available/