KLayout/klayout #1870: Shapes on certain layer not selectable

martinjankoehler, 2024-09-24:
Version: KLayout 0.29.7, commit 2d4f5d822c2e8c2649387ad5c5e242bbb85c56ec

I can't select shapes on `mcon.pin 78748`, also the icon shows an X:
[image]

Not sure if this is a bug or a PEBKAC, but if the latter, where can this behaviour be configured/toggled?

Demo Video:
https://github.com/user-attachments/assets/9f7166af-9457-443d-ab50-686a766f9795

GDS:
[klayout_unselectable_mcon-pin.gds.gz](https://github.com/user-attachments/files/17115981/klayout_unselectable_mcon-pin.gds.gz)

sebastian-goeldi, 2024-09-24:
Hello, I believe that's because the layer is not set as valid (there is a flag to make a layer valid/invalid). I am on mobile, so I cannot test. Could you test whether it works if you right click the layer and click "Make layer valid" (or similar name, can't remember the exact wording). This is a setting gotten from the .lyp file (or technology if that's what you use)

martinjankoehler, 2024-09-24:
@sebastian-goeldi Thanks, I can confirm it works exactly like you told!

@klayoutmatthias OK, then IMO KLayout should warn when somebody attempts to draw in an invalid layer,
perhaps similar like it already warns when somebody tries to draw without any layer selected:
[image]

sebastian-goeldi, 2024-09-24:
For selection this is the intended behavior.
https://klayout.de/doc/manual/layer_validity.html

I can't really speak for drawing, as I never use the GUI for drawing but rather the python/pypi standalone for drawing and the GUI for viewing. But if the GUI doesn't let you draw on a invalid layer, I think yes it would make sense for it to throw an error message like it does when no layer is selected.

martinjankoehler, 2024-09-24:
> But if the GUI doesn't let you draw on a invalid layer, I think yes it would make sense for it to throw an error message like it does when no layer is selected.

Yeah currently it actually allows drawing but not selecting, which to me does not seem consistent.
That's what happened in my use case, I wanted 'met1.pin' (valid) but somehow selected 'mcon.pin' (invalid),
and then accidentally drew a box which I was not able to remove again (at least via selection and delete key).

klayoutmatthias, 2024-09-24:
I see ...

Yes I agree. It never occured to me to draw on an invalid layer, but there should at least be some warning. Like when you try to draw on an invisible layer - which is equally confusing :)

Matthias

closed 2024-10-06 by klayoutmatthias