KLayout/klayout #1871: Mac DMGs for 0.29.7

Kazzz-S, 2024-09-27:
Dear @klayoutmatthias,

Thank you for the new maintenance release.
As usual, different DMG packages of **KLayout 0.29.7 for Mac** are available as described below ([0.29.7 Dropbox Link](https://www.dropbox.com/scl/fo/1itxzb78z9i5qq9ig2cic/ANqoWJZ837is2Z971ItcSx4?rlkey=wnmor6fgvg3mae5xeif5pru6q&dl=0)).
Please note that:
1. **Monterey (12.7.6)** has been dropped from the DMG table since **Homebrew** stopped supporting this OS.
2. **Sequoia (15.0)** builds will need more time.
3. **Python 3.12** is used in **LW-*** (earlier, 3.11).

## DMG file naming convention ##
[image]

where
(a) Type of package:  ST-:standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt[5|6] type:  qt[5|6]MP: MacPorts; qt[5|6]Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:   sys: OS-bundled; mp33: Ruby3.3 from MacPorts; hb33: Ruby3.3 from Homebrew; ana3: Ruby3.2 from Anaconda3
(f) **P**ython type: sys: OS-bundled; mp312: Python 3.12 from MacPorts; hb[312|311]: Python 3.[12|11] from Homebrew; ana3: Python3.12 from Anaconda3

## For the primary platform, macOS Sonoma (14.7) ##
Five macOS DMGs for **Sonoma** are available.

**Python is (3.9, 3.12, 3.[12|11], 3.12)=(OS-bundled, MacPorts, Homebrew, Anaconda3)**
**Ruby is (2.6, 3.3, 3.3, 3.2)=(OS-bundled, MacPorts, Homebrew, Anaconda3)**
|Srl.No.|Target OS|DMG file name|Qt5 or Qt6|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
10|Sonoma (14.7) |ST-klayout-0.29.7-macOS-Sonoma-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(10)|
11|Sonoma (14.7) with MacPorts dev. env.|LW-klayout-0.29.7-macOS-Sonoma-1-qt5MP-Rmp33Pmp312.dmg|MacPorts|MacPorts|MacPorts|(11)|
12|Sonoma (14.7) with Homebrew dev. env.|LW-klayout-0.29.7-macOS-Sonoma-1-qt6Brew-Rhb33Phb312.dmg|Homebrew|Homebrew|Homebrew|(12)|
13|Sonoma (14.7) with Anaconda3 dev. env.|LW-klayout-0.29.7-macOS-Sonoma-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(13)|
14|Sonoma (14.7)|HW-klayout-0.29.7-macOS-Sonoma-1-qt5MP-RsysPhb311.dmg|MacPorts|OS-bundled|Homebrew|(14)|

(10) shares OS-bundled Ruby2.6 and Python3.9 (dev. tools required); Qt5 from MacPorts is embedded.
(11) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(12) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(13) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(14) **experimental**; shares OS-bundled Ruby2.6; Qt5 from MacPorts and Python-3.11 from Homebrew are embedded, hence, heavy.

## For macOS Ventura (13.7) ##
Five macOS DMGs for **Ventura** are available.
They were built on an Intel-CPU-based (not Apple Silicon chip) VMware Fusion 13.6.0 on Sonoma (14.7).

**Python is (3.9, 3.12, 3.[12|11], 3.12)=(OS-bundled, MacPorts, Homebrew, Anaconda3)**
**Ruby is (2.6, 3.3, 3.3, 3.2)=(OS-bundled, MacPorts, Homebrew, Anaconda3)**
|Srl.No.|Target OS|DMG file name|Qt5 or Qt6|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
20|Ventura (13.7) |ST-klayout-0.29.7-macOS-Ventura-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(20)|
21|Ventura (13.7) with MacPorts dev. env.|LW-klayout-0.29.7-macOS-Ventura-1-qt5MP-Rmp33Pmp312.dmg|MacPorts|MacPorts|MacPorts|(21)|
22|Ventura (13.7) with Homebrew dev. env.|LW-klayout-0.29.7-macOS-Ventura-1-qt6Brew-Rhb33Phb312.dmg|Homebrew|Homebrew|Homebrew|(22)|
23|Ventura (13.7) with Anaconda3 dev. env.|LW-klayout-0.29.7-macOS-Ventura-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(23)|
24|Ventura (13.7)|HW-klayout-0.29.7-macOS-Ventura-1-qt5MP-RsysPhb311.dmg|MacPorts|OS-bundled|Homebrew|(24)|

(20) shares OS-bundled Ruby2.6 and Python3.9 (dev. tools required); Qt5 from MacPorts is embedded.
(21) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(22) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(23) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(24) **experimental**; shares OS-bundled Ruby2.6; Qt5 from MacPorts and Python-3.11 from Homebrew are embedded, hence, heavy.

## For the latest macOS Sequoia (15.0) ##
> [!NOTE]
> Initially, I planned to build KLayout 0.29.7 for the latest **Sequoia**.
> However, I am postponing the plan due to the low maturity of the OS I installed in the VMware Fusion 13.6.0 environment. So please ignore this section.
>
> The macOS build system will be updated shortly. If you have a real **Sequoia** machine, you may try to build KLayout using the source code.

Five macOS DMGs for **Sequoia** are available.
They were built on an Intel-CPU-based (not Apple Silicon chip) VMware Fusion 13.6.0 on Sonoma (14.7).

**Python is (3.9, 3.12, 3.[12|11], 3.12)=(OS-bundled, MacPorts, Homebrew, Anaconda3)**
**Ruby is (2.6, 3.3, 3.3, 3.2)=(OS-bundled, MacPorts, Homebrew, Anaconda3)**
|Srl.No.|Target OS|DMG file name|Qt5 or Qt6|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
30|Sequoia (15.0) |ST-klayout-0.29.7-macOS-Sequoia-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(30)|
31|Sequoia (15.0) with MacPorts dev. env.|LW-klayout-0.29.7-macOS-Sequoia-1-qt5MP-Rmp33Pmp311.dmg|MacPorts|MacPorts|MacPorts|(31)|
32|Sequoia (15.0) with Homebrew dev. env.|LW-klayout-0.29.7-macOS-Sequoia-1-qt6Brew-Rhb33Phb311.dmg|Homebrew|Homebrew|Homebrew|(32)|
33|Sequoia (15.0) with Anaconda3 dev. env.|LW-klayout-0.29.7-macOS-Sequoia-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(33)|
34|Sequoia (15.0)|HW-klayout-0.29.7-macOS-Sequoia-1-qt5MP-RsysPhb311.dmg|MacPorts|OS-bundled|Homebrew|(34)|

(30) shares OS-bundled Ruby2.6 and Python3.9 (dev. tools required); Qt5 from MacPorts is embedded.
(31) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(32) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(33) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(34) **experimental**; shares OS-bundled Ruby2.6; Qt5 from MacPorts and Python-3.11 from Homebrew are embedded, hence, heavy.

## For previous OS ##
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
4|Monterey (12.7.6)|From 0.29.7 onward, DMGs will not be provided for this OS.||||(4)|
5|Big Sur (11.7.10)|From 0.28.13 onward, DMGs will not be provided for this OS.||||(1p)|
6|Catalina (10.15.7)|From 0.28.13 onward, DMGs will not be provided for this OS.||||(1p)|
7|Mojave (10.14)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
8|HighSierra (10.13)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
9|Sierra (10.12)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
10|ElCapitan (10.11)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
x|Yosemite (10.10)|has been dropped due to the compiler issue reported in #360.|

(4) the build is still possible, but Homebrew stopped supporting this OS in September 2024.
(1p) From 0.28.13 onward, build is no longer possible.

Please close this ticket after publishing the DMGs.
With warm regards,
Kazzz-S

klayoutmatthias, 2024-09-28:
DMGs are published. Many thanks as usual! :)

closed 2024-09-28 by klayoutmatthias

Kazzz-S, 2024-10-03:
Hello @klayoutmatthias,

I created the following DMGs on **Sequoia**, which you can find here ([0.29.7 Dropbox Link](https://www.dropbox.com/scl/fo/1itxzb78z9i5qq9ig2cic/ANqoWJZ837is2Z971ItcSx4?rlkey=wnmor6fgvg3mae5xeif5pru6q&dl=0)).

## For the latest macOS Sequoia (15.0) ##
Five macOS DMGs for **Sequoia** are available.
They were built on an Intel-CPU-based (not Apple Silicon chip) VMware Fusion 13.6.0 on Sonoma (14.7).

**Python is (3.9, 3.12, 3.[12|11], 3.12)=(OS-bundled, MacPorts, Homebrew, Anaconda3)**
**Ruby is (2.6, 3.3, 3.3, 3.2)=(OS-bundled, MacPorts, Homebrew, Anaconda3)**
|Srl.No.|Target OS|DMG file name|Qt5 or Qt6|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
30|Sequoia (15.0) |ST-klayout-0.29.7-macOS-Sequoia-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(30)|
31|Sequoia (15.0) with MacPorts dev. env.|LW-klayout-0.29.7-macOS-Sequoia-1-qt5MP-Rmp33Pmp311.dmg|MacPorts|MacPorts|MacPorts|(31)|
32|Sequoia (15.0) with Homebrew dev. env.|LW-klayout-0.29.7-macOS-Sequoia-1-qt6Brew-Rhb33Phb311.dmg|Homebrew|Homebrew|Homebrew|(32)|
33|Sequoia (15.0) with Anaconda3 dev. env.|LW-klayout-0.29.7-macOS-Sequoia-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(33)|
34|Sequoia (15.0)|HW-klayout-0.29.7-macOS-Sequoia-1-qt5MP-RsysPhb311.dmg|MacPorts|OS-bundled|Homebrew|(34)|

(30) shares OS-bundled Ruby2.6 and Python3.9 (dev. tools required); Qt5 from MacPorts is embedded.
(31) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(32) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(33) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(34) **experimental**; shares OS-bundled Ruby2.6; Qt5 from MacPorts and Python-3.11 from Homebrew are embedded, hence, heavy.

Regards,
Kazzz-S

klayoutmatthias, 2024-10-03:
Many thanks! Files are on the server.

I would like to release a new version soon because of that stupid bug: https://github.com/KLayout/klayout/issues/1874.
I hope that does not create any inconvenience.

Best regards,

Matthias

Kazzz-S, 2024-10-03:
No worry.  :-)
I was able to upgrade my development environments through this release altogether.

Best regards,
Kazzz-S