KLayout/klayout #1873: klayout save as adds .o to .oas extension and refuses to save

stefanottili, 2024-09-28:
MacOS Sequoia MacBook M1

[image]

klayoutmatthias, 2024-09-28:
Is that specific to Sequioa maybe? As far as I understand, the current builds do not support that version yet.

Matthias

stefanottili, 2024-09-28:
I had this problem on the previous MacOS Sonoma too and compile klayout myself.

klayoutmatthias, 2024-09-28:
I can only guess that it is the system file dialog which kind of chokes on the file filters. 

In my case, the filter string is long, but appears to be correct:

```
All layout files (*.gds *.GDS *.gds.gz *.GDS.gz *.GDS2 *.gds2 *.gds2.gz *.GDS2.gz *.txt *.TXT  *.oas *.OAS *.oas.gz *.OAS.gz  *.lef *.LEF *.lef.gz *.LEF.gz *.def *.DEF *.def.gz *.DEF.gz *.pcb *.PCB *.dxf *.DXF *.dxf.gz *.DXF.gz *.cif *.CIF *.cif.gz *.CIF.gz *.mag *.MAG *.mag.gz *.MAG.gz);;GDS2 files (*.gds *.GDS *.gds.gz *.GDS.gz *.GDS2 *.gds2 *.gds2.gz *.GDS2.gz);;GDS2 Text files (*.txt *.TXT );;OASIS files (*.oas *.OAS *.oas.gz *.OAS.gz);;LEF/DEF files (*.lef *.LEF *.lef.gz *.LEF.gz *.def *.DEF *.def.gz *.DEF.gz);;Gerber PCB project files (*.pcb *.PCB);;DXF files (*.dxf *.DXF *.dxf.gz *.DXF.gz);;CIF files (*.cif *.CIF *.cif.gz *.CIF.gz);;Magic files (*.mag *.MAG *.mag.gz *.MAG.gz);;All files (*)
```

The function used from Qt is this: https://doc.qt.io/qt-6/qfiledialog.html#getSaveFileName

The documentation does not disagree with the filter string, the way I read it.

Matthias

Kazzz-S, 2024-09-29:
My main problem with **Sequoia** (15.x) is its combination with VMware Fusion.
For example, I cannot even select the right keyboard.
So, I'm not yet able to start installing the development tools.
A real machine may work much better, but I'm not brave enough to upgrade my main machine.
I only recently upgraded it from **Monterey** (12.x) to **Sonoma** (14.x).
I plan to test the fresh **Sequoia** installation with an external SSD (cost is a pain).

----

I tested `LW-klayout-0.29.7-macOS-Sonoma-1-qt6Brew-Rhb33Phb312.dmg` on my Intel Mac using Homebrew Qt 6.7.2 and
`Sava As OASIS` works normally.
[image]

Kazzz-S

stefanottili, 2024-10-20:
I’ll revisit this once a newer homebrew qt version gets released, both 6.7.3 and 6.8 are out there.

closed 2024-10-20 by stefanottili