KLayout/klayout #1884: KLayout macOS 29.7 Infinite window launch

Revenantx86, 2024-10-09:
Hello, 

I have recently installed klayout on my Mac and added sky130 libraries to do some layout. 
However, when I launch the klayout after adding the library, klayout won't stop launching new window and eventually crashing my laptop. Is there a way to fix this issue. I believe the issue is from the pcells script that running with python and its just keeps lauching new klayout window. I have installed all libraries required.

Thanks

klayoutmatthias, 2024-10-09:
I don't think this is a KLayout bug, so this may not be the right place to raise this question.

What exactly did you install? Specifically, from where and which version?

Matthias

Revenantx86, 2024-10-09:
I installed it using the DMG image from the website, and the SKY130 PDK is from the regular open_pdk repository. I added the SKY130 PDK using the technology package manager. Then I installed GDSFactory because the SKY130 library requires it for the PCells. However, to install GDSFactory, I used start-console.py since the Python version KLayout is using is different from the default version. After the installation, when I run the SKY130 scripts, it just keeps launching KLayout windows.  

I am posting here because the same setup on my Linux machine doesn't produce this error, so I thought it could be related to KLayout.

Refik

Kazzz-S, 2024-10-10:
Hello,

> [!IMPORTANT]
> Please provide more info on your Mac, including the OS name and DMG you installed.

I'm a complete novice to the `SKY130 PDK` project.
Please kindly guide me step-by-step to reproduce your problem.
I have done the following so far.

## On Linux Mint 20.3 ##
**1. Install Open-PDKs**
```
git clone https://github.com/RTimothyEdwards/open_pdks.git
cd open_pdks
./configure [options]
make
[sudo] make install
```

* In the `configure` stage, I got
```
checking for a Python interpreter with version >= 3.4... python3
checking for python3... /usr/local/bin/python3
checking for python3 version... 3.12
checking for python3 platform... linux
checking for python3 script directory... ${prefix}/lib/python3.12/site-packages
checking for python3 extension module directory... ${exec_prefix}/lib/python3.12/site-packages
checking python3 module: distutils... yes
checking for a sed that does not truncate output... /usr/bin/sed
configure: Found technology directories: gf180mcu sky130
Checking technology gf180mcu...
Checking technology sky130...
configure: Found tools: klayout magic netgen irsim openlane qflow xschem xcircuit
checking for magic... no
configure: error: You need 'magic' to generate the needed various cell views. <===
```

So I installed `magic` by:
```
$ sudo apt-get install magic

$ magic --version

8.2.157 <===
```

> [!IMPORTANT]
> Is the above version OK?
> How did you install `magic` on your Mac? From Homebrew?

```
checking for a Python interpreter with version >= 3.4... python3
checking for python3... /usr/local/bin/python3
checking for python3 version... 3.12
checking for python3 platform... linux
checking for python3 script directory... ${prefix}/lib/python3.12/site-packages
checking for python3 extension module directory... ${exec_prefix}/lib/python3.12/site-packages
checking python3 module: distutils... yes
checking for a sed that does not truncate output... /usr/bin/sed
configure: Found technology directories: gf180mcu sky130
Checking technology gf180mcu...
Checking technology sky130...
configure: Found tools: klayout magic netgen irsim openlane qflow xschem xcircuit
checking for magic... /usr/bin/magic
checking for patch... /usr/bin/patch
configure: Package 'sky130_fd_pr' will be installed automatically during make.
configure: Package 'sky130_fd_io' will be installed automatically during make.
configure: Package 'sky130_fd_sc_hs' will be installed automatically during make.
configure: Package 'sky130_fd_sc_ms' will be installed automatically during make.
configure: Package 'sky130_fd_sc_ls' will be installed automatically during make.
[... 98 lines cut ...]
* What's wrong? Am I doing the right thing?

**3. Install GDSFactory 8.12.0**
```
$ pip3 install gdsfactory

$ pip3 list | grep gds

gdsfactory           8.12.0 <===

(KLPy312) LinuxMint{sekigawa}(1)$ python3
Python 3.12.7+ (heads/3.12:79b4c9d99a0, Oct  8 2024, 08:16:52) [GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gdsfactory

>>> gdsfactory.__file__
'/home/sekigawa/opt/KLPy312/lib/python3.12/site-packages/gdsfactory/__init__.py' <=== using a virtual env.
>>>
```
[image]

> [!IMPORTANT]
> How do I integrate the Open-PDKs scripts into the virtual environment?

To be continued...

Kazzz-S, 2024-10-10:
Cont.

I successfully cloned the five repositories below using `SmartGit` on Linux.
1. https://github.com/FatsieFS/Salt-C4M_FlexLibs.git
2. https://github.com/FatsieFS/Salt-C4M_PDK_Sky130
3. https://github.com/FatsieFS/Salt-PDKMaster.git
4. https://github.com/klayoutmatthias/sky130A_el.git
5. https://github.com/klayoutmatthias/xsection.git

```
:
: my local git repos
:
 4 drwxrwxr-x  5 sekigawa sekigawa  4096 Oct 10 14:19 Salt-C4M_FlexLibs/
 4 drwxrwxr-x  7 sekigawa sekigawa  4096 Oct 10 14:20 Salt-C4M_PDK_Sky130/
 4 drwxrwxr-x  5 sekigawa sekigawa  4096 Oct 10 14:21 Salt-PDKMaster/
 4 drwxrwxr-x 10 sekigawa sekigawa  4096 Oct 10 14:25 sky130A_el/
 4 drwxrwxr-x  8 sekigawa sekigawa  4096 Oct 10 14:25 xsection/
 ```

Should I just create symbolic links to these directories from `$HOME/.klayout/salt/`?
If so, in what directory structure?

> [!NOTE]
> My KLayout 0.29.7 (source-code-build) on Linux uses `libgit2 1.8.1` (source-code-build).
> The Windows binary seems to link `libgit2 1.7`.

Revenantx86, 2024-10-10:
Hello, if you have installed the klayout on Mac can you try adding the sky130 library on the open_pdk. You can either build it form open_pdks or install quickly with efabless Volare. The existing packages on klayout are working on me as well but they are not complete, missing instances.

Kazzz-S, 2024-10-11:
Hello @Revenantx86,

As seen above, I have resolved the package installation problem on Linux.
I have installed:
1. Open-PDKs
2. SKY130 PDK
3. GDSFactory globally and locally (virtual environment); so either environment can be used

I believe my Linux setup is identical to yours.

> I am posting here because the same setup on my Linux machine doesn't produce this error,
> so I thought it could be related to KLayout.

The next step is to know your operation sequence on Linux, which `DOES NOT produce the error`.
In other words, your expected behaviors.
I will replay the same sequence on Linux first, then on Mac (I hope to try both next week).

> However, to install GDSFactory, I used **start-console.py** since the Python version KLayout is using
> is different from the default version.

From this description, I assume you are using one of `HW-*.dmg`, including Python 3.11.10. Please let me know if not.

> After the installation, when I run the SKY130 scripts, it just keeps launching KLayout windows.

Please specify which scripts you ran.

Regards,
Kazzz-S

klayoutmatthias, 2024-10-11:
Hi @Revenantx86,

before going too deep into debugging, you should try boiling down the problem.

My preferred way is use a fresh isolated environment. For Python that is a new virtual environment, for KLayout that is a new KLAYOUT_HOME.

First you need to identify the Python version you're using with the DMG. Under Linux that is always the system Python, under Mac there are different options. I am assuming you are using system Python.

To create a virtual environment I use:

```
python3 -m venv venv-test
. venv-test/bin/activate
```

If you a using a non-system Python with KLayout, replace "python3" by that Python interpreter.

Now you can install GDSFactory there. As you suggest, I am going to install "gf180mcu" which in the current 0.0.4 version is not compatible with GDSFactory 8. Hence I use version 7.27.1 which is the latest one of the 7 series:

```
pip3 install gdsfactory==7.27.1
```

To convince KLayout to use the virtual environment you need to point it to the virtual environment. First, you need to disable the system environment which the Python interpreter embedded into KLayout loads automatically. You can do so by setting `KLAYOUT_PYTHONHOME` to some non-existing directory, i.e. `/dev/null`. 

We also need to copy the Python search path into `KLAYOUT_PYTHONPATH`. 

So, inside the virtual environment do:

```
export KLAYOUT_PYTHONHOME=/dev/null
export KLAYOUT_PYTHONPATH=$(python3 -c "import sys; print(':'.join(sys.path))")
```

To use a fresh KLayout environment, you can use a new folder for KLAYOUT_HOME. For example:

```
rm -rf klayout-test
export KLAYOUT_HOME=$(pwd)/klayout-test
```

If you start KLayout now from the command line, you should see a blank environment. No packages loaded. 

You can now install "gf180mcu" from the package manager in version 0.0.4. Exit KLayout and restart. Now GDSFactory will complete it's installation by adding some more packages. Make sure you do all of that from the virtual environment.

I am now able to draw layouts with KLayout using the GF180 device primitives. There is no effect of windows popping up on Linux. So far that is consistent.

Maybe you can repeat the same steps on MacOS to see if that recipe allows reproducing the problem.

Matthias

Kazzz-S, 2024-10-15:
### Linux Mint 20.3 ###

Hello @klayoutmatthias,

Thank you for the instructions.
On Linux Mint 20.3, I usually build KLayout from source code using Python 3.12.7+, also built from source code and installed in `/usr/local/[bin|lib]`.
Below are the steps I followed.

**1. Create a virtual environment**
```
Mint203{sekigawa}(1)$ python3 -m venv $HOME/opt/KLPy312
Mint203{sekigawa}(2)$ source $HOME/opt/KLPy312/bin/activate
```

**2. Install "gdsfactory"**
```
(KLPy312) Mint203{sekigawa}(3)$ pip3 install gdsfactory==7.27.1

(KLPy312) Mint203{sekigawa}(4)$ pip3 list | grep factory
gdsfactory             7.27.1
kfactory               0.13.3
```

**3. Use a fresh KLayout environment**
```
(KLPy312) Mint203{sekigawa}(11)$ export VIRTUAL_ENV=$HOME/opt/KLPy312
(KLPy312) Mint203{sekigawa}(12)$ export KLAYOUT_PYTHONHOME=/dev/null
(KLPy312) Mint203{sekigawa}(13)$ export KLAYOUT_PYTHONPATH=$($VIRTUAL_ENV/bin/python3 -c "import sys; print(':'.join(sys.path))")

(KLPy312) Mint203{sekigawa}(14)$ \rm -rf klayout-test
(KLPy312) Mint203{sekigawa}(15)$ export KLAYOUT_HOME=$(pwd)/klayout-test
(KLPy312) Mint203{sekigawa}(16)$ klayout ===> started normally!
```

**4. Install "gf180mcu" from the package manager in version 0.0.4**

**5. Draw layouts with KLayout using the GF180 device primitives**
[image]

I have confirmed that multi-window popups do not occur on Linux in this case.
I will be conducting similar tests on Mac.

Best regards,
Kazzz-S

Kazzz-S, 2024-10-15:
### macOS Sonoma with LW-klayout-0.29.7-macOS-Sonoma-1-qt6Brew-Rhb33Phb312.dmg ###

**0. Install the DMG**
LW-klayout-0.29.7-macOS-Sonoma-1-qt6Brew-Rhb33Phb312.dmg --- install ---> /Applications/klayout.app

**1. Create a virtual environment**
```
MacBookPro2{sekigawa} ~ (1)% which python3
/usr/local/opt/python@3.12/bin/python3 <=== Python 3.12.7 from Homebrew

MacBookPro2{sekigawa} ~ (2)% python3 -m venv $HOME/opt/HBPy312
MacBookPro2{sekigawa} ~ (3)% source $HOME/opt/HBPy312/bin/activate
```

**2. Install "gdsfactory"**
```
(HBPy312) MacBookPro2{sekigawa} ~ (4)% which pip3
/Users/sekigawa/opt/HBPy312/bin/pip3

(HBPy312) MacBookPro2{sekigawa} ~ (5)% pip3 list
Package Version
------- -------
pip     24.2

(HBPy312) MacBookPro2{sekigawa} ~ (6)% pip3 install gdsfactory==7.27.1

(HBPy312) MacBookPro2{sekigawa} ~ (7)% pip3 list | grep factory
gdsfactory             7.27.1
kfactory               0.13.3
```

**3. Use a fresh KLayout environment**
```
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (10)% pwd
/Users/sekigawa/GitWork/ForumKLayout/GitHub/GitHub1884/macOS-Sonoma-LW

(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (11)% export VIRTUAL_ENV=$HOME/opt/HBPy312
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (12)% export KLAYOUT_PYTHONHOME=/dev/null
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (13)% export KLAYOUT_PYTHONPATH=$($VIRTUAL_ENV/bin/python3 -c "import sys; print(':'.join(sys.path))")

(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (14)% \rm -rf klayout-test
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (15)% export KLAYOUT_HOME=$(pwd)/klayout-test
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (16)% open -a /Applications/klayout.app ===> started normally!
```

**4. Install "gf180mcu" from the package manager in version 0.0.4**

**5. Draw layouts with KLayout using the GF180 device primitives**
[image]
```
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (17)% pip3 list | grep factory
<!!! EMPTY !!!>
```

**6. Re-install "gdsfactory==7.27.1" and restart the fresh KLayout**
```
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (18)% pip3 install gdsfactory==7.27.1
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (19)% pip3 list | grep factory
gdsfactory             7.27.1
kfactory               0.13.3

(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (20)% /Applications/klayout.app/Contents/MacOS/klayout
:
:
Using cached ruamel.yaml.string-0.1.1-py3-none-any.whl (4.1 kB)
Installing collected packages: rectangle-packer, attrs, ruamel.yaml.string, kfactory, gdsfactory
  Attempting uninstall: kfactory
    Found existing installation: kfactory 0.13.3
    Uninstalling kfactory-0.13.3:
      Successfully uninstalled kfactory-0.13.3 <=== ??? Why ???
  Attempting uninstall: gdsfactory
    Found existing installation: gdsfactory 7.27.1
    Uninstalling gdsfactory-7.27.1:
      Successfully uninstalled gdsfactory-7.27.1 <=== ??? Why ???
Successfully installed attrs-24.2.0 gdsfactory-8.13.3 kfactory-0.21.0 rectangle-packer-2.0.2 ruamel.yaml.string-0.1.1
===> ERROR: /Users/sekigawa/GitWork/ForumKLayout/GitHub/GitHub1884/macOS-Sonoma-LW/klayout-test/salt/gf180mcu/pymacros/cells/draw_fet.py:22: ModuleNotFoundError: No module named 'gdsfactory'
  /Users/sekigawa/GitWork/ForumKLayout/GitHub/GitHub1884/macOS-Sonoma-LW/klayout-test/salt/gf180mcu/pymacros/cells/draw_fet.py:22
  /Users/sekigawa/GitWork/ForumKLayout/GitHub/GitHub1884/macOS-Sonoma-LW/klayout-test/salt/gf180mcu/pymacros/cells/fet.py:19
  /Users/sekigawa/GitWork/ForumKLayout/GitHub/GitHub1884/macOS-Sonoma-LW/klayout-test/salt/gf180mcu/pymacros/cells/__init__.py:21
  /Users/sekigawa/GitWork/ForumKLayout/GitHub/GitHub1884/macOS-Sonoma-LW/klayout-test/salt/gf180mcu/pymacros/gf180mcu.lym:12 (class ModuleNotFoundError)

(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (21)% pip3 list | grep factory
<!!! EMPTY !!!>
```

**7. Re-install "gdsfactory(==8.13.3)" and restart the fresh KLayout**
```
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (31)% pip3 install gdsfactory
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (32)% pip3 list | grep factory
gdsfactory             8.13.3
kfactory               0.21.0
:
:
(HBPy312) MacBookPro2{sekigawa} macOS-Sonoma-LW (33)% /Applications/klayout.app/Contents/MacOS/klayout
```
* I think I was able to reproduce the original problem.

https://github.com/user-attachments/assets/22f4a4bf-f4e6-4286-b463-0ac6f1386a72

Best regards,
Kazzz-S

klayoutmatthias, 2024-10-15:
Hi @Kazzz-S Thanks for the analysis!

From what I see there is not much inside that file:

```
import sys
import os
import pip

pip.main(['install', 'gdsfactory', '--upgrade'])

technology_macros_path = os.path.dirname(os.path.abspath(__file__))
sys.path.insert(0, technology_macros_path)

from cells import gf180mcu

# Instantiate and register the library
gf180mcu()

print("## gf180mcu PDK Pcells loaded.")
print(sys.path)
```

The only thing which is kind of suspicious is the "pip install" call. Could you try commenting it out, like this:

```
# pip.main(['install', 'gdsfactory', '--upgrade'])
```

I even think that is a general problem as gf180mcu is pinned to an older version of gdsfactory.

Does the problem occur again if you set "autorun" back to "true"? If not, I must conclude that "pip install gdsfactory --upgrade" does some strange things which in the end call KLayout on MacOS only.

Best regards,

Matthias

klayoutmatthias, 2024-10-15:
I should add that for some reason gf180mcu needs an older version of gdsfactory. I assume their API evolved in an incompatible way. So you need to install 7.27.1 to run gf180mcu.

The best I guess was to update gf180mcu to support gdsfactory 8 too.

Matthias

Kazzz-S, 2024-10-21:
Cont.

I started KLayout by `/Applications/klayout-HB.app/Contents/MacOS/klayout -d 100 2>&1 | tee d100.log`.
Then, `import gdsfactory` was used as shown in **Direct import in Homebrew** above.
The log message tells that...
```
:
:
/usr/local/Cellar/python@3.12/3.12.7_1/Frameworks/Python.framework/Versions/3.12/\
lib/python3.12/multiprocessing/resource_tracker.py:123: \
UserWarning: resource_tracker: process died unexpectedly, relaunching.  Some resources might leak.
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  warnings.warn('resource_tracker: process died unexpectedly, '
```

I think `this` AND `autorun==true` are the root causes of this infinite-launching problem, not a KLayout issue.
Is `gdsfactory` designed for macOS to begin with?

Best regards,
Kazzz-S
[d-100.zip](https://github.com/user-attachments/files/17459363/d-100.zip)

martinjankoehler, 2025-06-06:
OK, to investigate this issue further, I've created a file `/Users/martin/bin/klayout-inf-loop.py`:

```python
#-----------------------------------------------------------
import subprocess
import os
import pdb

#-----------------------------------------------------------

_original_popen = subprocess.Popen

class DebugPopen(subprocess.Popen):
    def __init__(self, *args, **kwargs):
        print("[DebugPopen] Command:", args[0])
        if "klayout" in str(args[0]):
            pdb.set_trace()
        super().__init__(*args, **kwargs)

subprocess.Popen = DebugPopen
#-----------------------------------------------------------

_original_system = os.system
def debug_system(cmd):
    print("[debug_system]", cmd)
    return _original_system(cmd)
os.system = debug_system

#-----------------------------------------------------------
_original_execv = os.execv
def debug_execv(path, args):
    print("[debug_execv]", path, args)
    return _original_execv(path, args)
os.execv = debug_execv
#-----------------------------------------------------------

import gdsfactory
```

Then in the Terminal I start:
```bash
( klayout -rx -b -r ~/bin/klayout-inf-loop.py & )  ; sleep 5; killall -STOP klayout

# In the meantime, 3 klayout instances where created.

ps -o ppid,command | grep klayout
```

Here I get 
```bash
99078 /bin/bash /Users/martin/bin/klayout -rx
99078 /bin/bash /Users/martin/bin/klayout -rx -b -r /Users/martin/bin/klayout-inf-loop.py
    1 /bin/bash /Users/martin/bin/klayout -rx -b -r /Users/martin/bin/klayout-inf-loop.py
38859 /Users/martin/Source/klayout/qt5MP.build.macos-Sequoia-release-Rhb34Phbauto.macQAT/klayout.app/Contents/MacOS/klayout -rx -b -r /Users/martin/bin/klayout-inf-loop.py
38861 /Users/martin/Source/klayout/qt5MP.build.macos-Sequoia-release-Rhb34Phbauto.macQAT/klayout.app/Contents/MacOS/klayout -c from multiprocessing.resource_tracker import main;main(12)
38864 /Users/martin/Source/klayout/qt5MP.build.macos-Sequoia-release-Rhb34Phbauto.macQAT/klayout.app/Contents/MacOS/klayout -c from multiprocessing.resource_tracker import main;main(25)
38877 /Users/martin/Source/klayout/qt5MP.build.macos-Sequoia-release-Rhb34Phbauto.macQAT/klayout.app/Contents/MacOS/klayout -c from multiprocessing.resource_tracker import main;main(24)
```

So somebody calls
```bash
$> pstree 38864
-+- 38864 martin /Users/martin/Source/klayout/qt5MP.build.macos-Sequoia-release-Rhb34Phbauto.macQAT/klayout.app/Contents/MacOS/klayout -c from multiprocessing.resource_tracker import main;main(12)
 \-+- 38877 martin /Users/martin/Source/klayout/qt5MP.build.macos-Sequoia-release-Rhb34Phbauto.macQAT/klayout.app/Contents/MacOS/klayout -c from multiprocessing.resource_tracker import main;main(25)
   \--- 38889 martin /Users/martin/Source/klayout/qt5MP.build.macos-Sequoia-release-Rhb34Phbauto.macQAT/klayout.app/Contents/MacOS/klayout -c from multiprocessing.resource_tracker import main;main(24)```