KLayout/klayout #1885: how to specify an empty string for strm2oas --subst-char ?

stefanottili, 2024-10-10:
In klayout, if the substitution string is "empty" illegal characters in TEXT will not be substituted.

[image]

How can I specify an empty string for strm2oas on the command line ?

strm2oas --subst-char    test_illegal_char/test.oas a.oas 
ERROR: Additional arguments required (use -h for help)

strm2oas --subst-char "" test_illegal_char/test.oas a.oas
ERROR: Additional arguments required (use -h for help)

strm2oas --subst-char '' test_illegal_char/test.oas a.oas
ERROR: Additional arguments required (use -h for help)

strm2oas --subst-char \"\" test_illegal_char/test.oas a.oas
This is the only one not giving an ERROR, but uses * as substitution character.
TEXTSTRING "illegal*"

I had to gzip the .oas file to be able to attach it, please gunzip, it's a strict oas.
[test.oas.gz](https://github.com/user-attachments/files/17331250/test.oas.gz)

stefanottili, 2024-10-11:
I would prefer the "default" to be "don't replace anything", one could use --subst-char "*" to get the current behavior.
A warning about "illegal" characters would be nice either way.

klayoutmatthias, 2024-10-11:
Hi Stefan,

by "replacing with empty string" you mean to delete illegal chars entirely? I think this is not supported as of now. 

The "don't substitute" default behavior proposal is fine with me as long as it applies to "strm2oas". For the GUI application I think, the default should be some safe value. But it could be made optional, so you can turn off character substitution.

Matthias

stefanottili, 2024-10-11:
I always expect any tool to write exactly what it read by default.
That even applies to “illegal” oasis, garbage in garbage out.
Especially since “other tools” aka the $$$ tools allow one do this too.

Current I can’t switch off strm2oas subst-char, in klayout I can do that.

I believe the best solution is to change strm2oas default to “don’t substitute”.
One can use —subst-char “*” if one wants to substitute. A warning would be nice.

stefanottili, 2024-10-11:
gds doesn't have the concept of "illegal characters".
I've never seen them used in cell names, only in TEXT, they won't really do any harm there.

They were there either because a certain layout tools allows <tab> or <cr> to be entered in their text entry form and people created multi-line text entries.
Or because a certain fab used binary numbers in the layer:datatype 63:63 IPTAG TEXT as some sort of signature in their stdcell gds.

klayoutmatthias, 2024-10-12:
I changed the parser to accept empty --subst-char using

```
strm2oas --subst-char= in.gds out.oas
```

However, the default is "don't substitute" now, so this patch actually is not required. I left it for completeness.

Matthias

commit 19b2aea4 (2024-10-19): Fixed #1885

Now, the default of --subst-char in strm2oas is "don't substitute".
You can explicitly specify "don't substitute" using "--subst-char="
(without value).
commit 9cf3c528 (2024-10-19): Merge pull request #1888 from KLayout/issue-1885

Fixed #1885
closed 2024-10-19 by klayoutmatthias