KLayout/klayout #1892: read lef/def feature request: issue WARNING if VIA SPACING is an odd number

stefanottili, 2024-10-12:
This is a feature request triggered by https://github.com/KLayout/klayout/issues/1877

When reading lef, an odd number for VIARULE..LAYER..SPACING will lead to rounding issues.
That's because via geometry has to be centered and thus there will be a divide by two used in the calculation of the anchor point of the top/bottom rectangle.

Issue a WARNING if the lef VIARULE..LAYER..SPACING is an odd number.
Issue a WARNING if the def VIA + CUTSPACING is an odd number.

stefanottili, 2024-10-13:
come to think about it, just issue a WARNING if there division by two of an odd number when reading the def VIA.

klayoutmatthias, 2024-10-14:
Yes, I agree, there should be a warning. Basically, it's the via array bounding box which should not have a odd width or height. 

I'd prefer to fix this along with #1877 because there is already a test case.

Thanks,

Matthias

klayoutmatthias, 2024-10-14:
I have included the warning into the #1877 PR.

closed 2024-10-19 by stefanottili