KLayout/klayout #1897: pya.Application.instance().main_window().menu().insert_menu -- change in behaviour?

lukasc-ubc, 2024-10-17:
Hi Matthias,

I notice that the custom menu I added doesn't show up in the display in the newer version.  Is this an intentional change, and is there a way to have it appear like it did in the 0.29.1, as per below?

KLayout 0.29.1 menu bar:
[image]

KLayout 0.29.6 menu bar:
[image]

This is implemented using:
> pya.Application.instance().main_window().menu().insert_menu("help_menu", "siepic_menu", "SiEPIC 0.5.13 Py3")

I checked the documentation and couldn't find anything suggesting this should not be possible: https://www.klayout.de/doc-qt5/code/class_AbstractMenu.html

thank you

Kazzz-S, 2024-10-17:
Hello @lukasc-ubc,

The behavior changed in `0.29.2`, which is macOS-specific.
[image]

Document ===> https://www.klayout.de/doc-qt5/about/macro_in_menu.html

[image]

The report that prompted the fix ===> https://github.com/KLayout/klayout/issues/1722

lukasc-ubc, 2024-10-17:
Thank you. Given that it worked previously, would there be a way to return that functionality? Or is it a major change in Qt that prevents this moving forward?

klayoutmatthias, 2024-10-18:
Hi @Kazzz-S - thanks for the pointer. I already forgot that issue ...

@lukasc-ubc Problem was that it was possible to add menu items, but when they get removed (which is a valid operation), KLayout crashed on MacOS. The "Extra" menu (with the three dots) as a container for dynamic content was the only feasible solution I found during long debugging sessions via a remote connection. I don't know if it is a Qt bug or some more general limitation, but it appears that it is not possible to dynamically reconfigure the application's top level menu - at least when it comes to removing items.

Matthias

lukasc-ubc, 2024-10-19:
Perhaps instead, 
1) disable the “remove” option for Mac OSX.
2) leave the “…” Extra menu in case people want to dynamically remove things.  Or those that want to remove, can themselves create a “…” menu item.

I suspect adding to the menu is a much more common operation than removing from the menu?  

Thank you

klayoutmatthias, 2024-10-19:
There are actually a number of practical issues. Thing is, dynamic menus work perfectly fine on other platforms, so this feature is being used. 

I really spent a lot of time trying to bring the behavior to MacOS including attempts to disable them instead of removing them. I could not get it stable.

Menus are dynamic not only because they are explicitly removed, but also when you remove macros or change a macro's menu binding for.example. So it's not enough to simply disable "remove".

Maybe someone with a deeper insight into.Qt's MacOS binding than me could take a look. If there is a way to bring the original behavior to MacOS in a stable way, I'll be happy to adopt that solution.

Matthias