KLayout/klayout #1898: Automated Salt submission via GitHub Action

lukasc-ubc, 2024-10-17:
Hi Matthias,

I have a GitHub Action that automatically publishes to PyPI: https://github.com/SiEPIC/SiEPIC_EBeam_PDK/blob/master/.github/workflows/python-publish.yml

I am wondering if it would be possible to automate the submission to your Salt server, via the https://sami.klayout.org/register page?  The URL can be used to pre-fill the form, but is there a way to have the button get pushed automatically via curl or some other method?

I would imagine receiving the confirmation email, which would be simple to click on it to accept.

thank you

klayoutmatthias, 2024-10-18:
Hi Lukas, 

that should be possible. The request is a simple GET.

With user = "myself", mail = "xzy@abd.de", provider = "Github" (==5), Project URL = "blabla" and "yes I agree to the terms" this URL is:

```
https://sami.klayout.org/register_request?user=myself&mail=xyz%40abc.de&provider=5&url_path=blabla&agree_terms=1
```

It's probably not wise to disclose this publicly, but these are details that you can easily trace for example with FireFox's network monitor.

Matthias0

lukasc-ubc, 2024-10-28:
I got it working.  Thank you very much.   All I had to do is click on the approval email that your system sent.

For future reference, here is my action file: https://github.com/SiEPIC/SiEPIC_EBeam_PDK/blob/master/.github/workflows/python-publish.yml