KLayout/klayout #1899: QPoint arithmentic operators problem

EugeneZelenko, 2024-10-18:
Hi, Matthias!

I experienced next problem after upgrading KLayout/Python/Qt: `RuntimeError: TypeError: unsupported operand type(s) for -: 'QPoint' and 'QPoint' in QFrame.contextMenuEvent`

Code snippet:

```
def contextMenuEvent(self, event):
        pos = event.pos() - pya.QPoint(0, self.__table_widget.horizontalHeader.height)
```

https://doc.qt.io/qt-6/qpoint.html list this and other arithmetic operators overloads as valid. Same may be true for other operators and primitive classes.

Build where this code worked: KLayout 0.28.12, Python 3.11.4, Qt 5.14.2. My current build uses KLayout 0.29.7, Python 3.12.7, Qt 6.7.3.

klayoutmatthias, 2024-10-18:
Yes, you're right. Problem is here, that these operators are defined inside the QPoint class body for Qt6, while they are defined outside (and only as friends inside) the class body in Qt5. That screws up the Qt header parser (which is a mess anyway).

"operator-=" is available also in Qt6. As a workaround you should be able to use:

```
pos = event.pos()
pos -= pya.QPoint(0, self.__table_widget.horizontalHeader.height)
```

Matthias

EugeneZelenko, 2024-10-18:
Hi, Matthias!

I used `pos.y -= self.__table_widget.horizontalHeader.height` as workaround.

klayoutmatthias, 2024-10-18:
Even easier :) 

I will check if I can add the other operators.

Matthias

closed 2024-10-20 by klayoutmatthias

EugeneZelenko, 2024-10-22:
Hi, Matthias!

Thank you for so quick fix!