KLayout/klayout #1901: feature request: create an option to have strmxor generate results xor'ing with "empty" layers

stefanottili, 2024-10-18:
This is a feature request to have strmxor behave the same way as "all the other lvl tools out there".

If one file has a layer:datatype that's "empty" in the other file, strmxor by default will report the layer as missing to stdout. But it won't output any geometry into the result file. This is in contrast to all the other layers, which will get results. So the result file by itself is not sufficient to "see" what changed when loading it into klayout with the -m option.

The strmxor -l option will write the flattened geometry into the result file.

I would like strmxor to gain an option for the 3rd possible behavior and enable it by default:
Run "scanline xor" against the "empty" 2nd layer:datatype, effectively running an "or" and output that as the result.

This is computationally the most expensive option, but is what all other lvl tools do by default. And thus it makes it easier to compare e.g. the number of result polygon per layer.

klayoutmatthias, 2024-10-19:
Actually, with -l, the implementation is like that, but the non-merging behavior is a consequence of the internal optimization: if a polygon set is XOR'd against an empty one, it is simply copied. It can be turned off ("strict mode"), but there is no option available for this for the strmxor too.

stefanottili, 2024-10-19:
I don’t get the last sentence, ..

> It can be turned off ("strict mode")
what does strict (oasis) mode have to do with scanline  xor ?

> but there is no option available for this for the strmxor too.
Well there is currently no option, I know.
But it should be possible to add an option to strmxor to feed the flattened polygon into the scanline instead of bypassing it, no ?

klayoutmatthias, 2024-10-19:
Sorry, that was confusing. I referred to "boolean strict handling" (https://www.klayout.de/doc-qt5/code/class_Region.html#method259).

klayoutmatthias, 2024-10-19:
That is turned off in strmxor and there is no user option to turn it on for now.