KLayout/klayout #1902: Address Sanitizer Crash after editing and moving selection (MacOS)

martinjankoehler, 2024-10-18:
Just selected a couple of Polygons+Labels+Pins and then called `Edit > Selection > Move By`.
Trace attached below.

[Log at 2024-10-18 5-58-19 PM.txt](https://github.com/user-attachments/files/17438080/Log.at.2024-10-18.5-58-19.PM.txt)

[image]

martinjankoehler, 2024-10-18:
Another AddressSanitizer Crash, now just by moving viewport using middle mouse button.

[Log at 2024-10-18 6-44-48 PM.txt](https://github.com/user-attachments/files/17438426/Log.at.2024-10-18.6-44-48.PM.txt)

klayoutmatthias, 2024-10-19:
The first one was real. 

The second one I'm not sure. I can't reproduce it on gcc and maybe it's related to clang optimization or their specific STL's list implementation (I doubt so). I have pushed a commit to bugfix/issue-1902 branch that maybe fixes the problem. But I used that idiom a number of times, so there is some risk the problem appears in other places as well.

closed 2024-10-20 by klayoutmatthias