KLayout/klayout #1913: PCell cell.name= assignment for export to GDS

lukasc-ubc, 2024-10-27:
Hi Matthias,

I would like to save to a GDS/OAS file with PCells, using
> save_options.write_context_info=False  

and having the cell name be the display_text_impl(self) name, rather than the cell.basic_name

This way, I won't have things like Cell$1, Cell$2, etc, but rather the specific variant (where the cell parameters can go)

In the docs, the purpose of the displau_text_impl, is
>    https://www.klayout.de/doc-qt5/programming/ruby_pcells.html#k_2
>   KLayout will call this method to fetch a formatted string that represents the content of a PCell. This text is used for the cell tree and cell box labels. 

I can change the cell.name after I instantiate the PCell, by finding the instance.  But within the PCell implementation code, a self.cell.name assignment doesn't work.

thank you

klayoutmatthias, 2024-10-27:
Hi Lukas,

there are a number of problems with that proposal. First, the display name really is a display name, nothing else. There is no warranty it is unique (needed for GDS), nor does it satisfy cell name rules such as allowed characters. In the end it's the same thing that Cadence does, when it streams out GDS and attaches some hash value to the cell name.

You can however propose your own name when you create a PCell. Like in this example which uses the display title for the name. But use this scheme at your own risk. The GDS reader will still replace blanks by "$" because those are definitely among the forbidden characters.

```
import pya

cell = pya.CellView.active().cell
ly = cell.layout()
layer = ly.layer(1, 0)

# The parameter names are specific for the 
parameters = {
  "layer": pya.LayerInfo(1, 0),
  "text": "SOME TEXT FOR THE QR CODE",
  "pixel_size": 10.0,
  "bias": 0.0
}

qrcode_cell = ly.create_cell("QRCode", "QRCodeLib", parameters)

# use the display title as your own name for the cell
qrcode_cell.name = qrcode_cell.display_title()

xpos = 0.0
ypos = 0.0

cell.insert(pya.CellInstArray(qrcode_cell.cell_index(), pya.DTrans(xpos, ypos)))
```

Matthias

klayoutmatthias, 2024-10-27:
Just some more options:

Use this custom query to do the assignment of display title before saving:

```
with cells * do cell.name = cell.display_title
```

or with replacing blanks by "nothing":

```
with cells * do cell.name = gsub(cell.display_title, " ", "")
```

And for the cross-reference you can use

```
select cell.name, cell.display_title from cells *
```

Matthias

lukasc-ubc, 2024-10-28:
Thanks Matthias.  

I am successful in change the name, after instantiating the pcell code, as you write above.

I was originally hoping to do the name assignment WITHIN the PCell code, rather than outside.  as in "self.cell.name = 'xxx'"   But I can survive without.

thank you