KLayout/klayout #1915: .pyi files for pya module

EugeneZelenko, 2024-10-30:
Hi, Matthias!

It'll be great to have Python interface files for `pya` module (for both KLayout and Qt bindings). It'll help to enforce type validation with MyPy.

Sure, it would be huge task.

sebastian-goeldi, 2024-10-31:
They already exist, but I think only for the python standalone module. Not sure how much work it would be to have them available in the GUI version

klayoutmatthias, 2024-10-31:
The main problem is where to install them. The built in pya module does not have an installation space. It is just there, because this modules is bootstrap-loaded when KLayout initializes the interpreter.

Maybe there is a way to "borrow" them for PyCharm?

Matthias

sebastian-goeldi, 2024-10-31:
@klayoutmatthias I believe you can make a [stub only package](https://peps.python.org/pep-0561/#stub-only-packages) and just include it in the library folder. I believe it then should be called `pya-stubs` and any python typechecker should be able to pick it up. I once did this manually for klayout when it didn't have automatic type files for the standalone yet.

I'd be happy to help if there are problems or even just test it.

klayoutmatthias, 2024-10-31:
Okay, that sounds easy. I will try if I can create such a module.