KLayout/klayout #1922: missing windows wheels for pyhton3.13

joamatab, 2024-11-10:
How can we build Python wheels for Windows on Python 3.13?

Some GDSFactory users are starting to use Python 3.13, but pip install klayout fails for this version. Here’s the relevant GitHub Actions log for reference:
[GitHub Actions Log](https://github.com/gdsfactory/gdsfactory/actions/runs/11767515905/job/32776273449)

KLayout package files on PyPI:
[KLayout on PyPI](https://pypi.org/project/klayout/#files)

klayoutmatthias, 2024-11-12:
Thanks for this note. 3.13 is simply not included in the azure pipelines. I'll add it and let you know if it works. 

Matthias

klayoutmatthias, 2024-11-12:
Not that easy unfortunately as they deprecated Py_SetPath in 3.13.

They changed the logic of the initialization, so the code has to be restructured for 3.13. Which is a pain, because there are already #if's for telling Python 2 from 3 and Windows from Linux.

So overall not a simple change.

Matthias

klayoutmatthias, 2024-11-12:
I am trying a simple hack now ...

klayoutmatthias, 2024-11-13:
I think it worked now. There are Python 3.13 packages for amd64 and x86.

Could you try and let me know if they work?

Thanks,

Matthias

joamatab, 2024-11-15:
yes, Thank you Matthias, it worked great for windows on python3.13 now

https://github.com/gdsfactory/gdsfactory/actions/runs/11861799641/job/33059860651

closed 2024-11-15 by joamatab