KLayout/klayout #1926: Add a mechanism to specify/change keybindings

hpretl, 2024-11-16:
As keybindings are a very personal choice, and also **super important** for efficient layout work, a mechanism to specify keybindings via an easy-to-construct configuration file would be awesome. Changing keybindings by using the built-in macro possibilities (in Ruby or Python) is definitely possible, but not every KLayout is or should be skilled enough to dive into this just for adapting keybindings.

All the keybindings are stored in one gigantic entry in `klayoutrc`, which makes managing or changing keybindings difficult. Further, people will want to use keybindings across different PDKs, so a setting/mechanism buried inside PDK files is also not ideal.

A proposal how this might work:
By setting `KLAYOUT_KEYBINDING` to point to an XML (or similar) file, in this file keybindings to one's liking can be defined, which are read during KLayout startup, and overwrite whatever keybindings are specified elsewhere.

sergeiandreyev, 2024-11-17:
I agree that it should be PDK-agnostic..
in the KLayout GUI there is `Setup->Application->Customize Menu` where the user can specify key bindings to his liking (and this will be stored in his `.klayoutrc` as I understand)
do you think it's not convenient/comfortable?

hpretl, 2024-11-17:
The question is how to set this initially. In VM usage scenarios, the home directory (and this the `klayoutrc` are not permanent, so a nice way to set the keybindings would be nice.

One alternative scenario that I have in mind is that I define keybindings in the KLayout GUI, and use the resulting `klayoutrc` as a starting point in the VM image. But I don't find this really clean.

hpretl, 2024-11-18:
If there are no other ideas or concerns, I propose to close this issue, and go with the "provide a proper `klayoutrc` somewhere" option. 

@klayoutmatthias What do you think?

hpretl, 2024-11-18:
@klayoutmatthias Is there a way to attach keybindings to the actions in the layers window, like "toggle visibility", "show", etc?

klayoutmatthias, 2024-11-19:
Yes, there is :)

Below in the list of menu items, there is a section called "Layer Panel Context Menu". Here I bound "Show all" to the "!" key (or rather Shift+1):

[image]

Best regards,

Matthias

hpretl, 2024-11-19:
@klayoutmatthias Excellent, thank you! I already added a few keystrokes, this will help 😀

hpretl, 2024-12-11:
Closing this issue, it works nicely via `klayoutrc`.

closed 2024-12-11 by hpretl