KLayout/klayout #1934: selecting, little improfments

olisnr, 2024-11-25:
i would like some changes:
1. a move-mode that only moves already selected things.
2. a change after cntr-V or B to this move mode
3. a selection-rectangle, that only select inside, but is modifiable to also select touching, may be per modifier key shift or so, or via the direction how the rectangle is drawn (left-right/up-down or reverse).
4. the option to filter cells that dont fit in the current view out of the selection

klayoutmatthias, 2024-11-26:
Hi @olisnr,

I think this refers to this discussion: https://www.klayout.de/forum/discussion/2624/how-to-lock-cells

regarding 1: if you have selected things, move only selects those. 

regarding 2: In File/Setup, under Application/Customize Menu you find at the bottom of the list of functions some "Key binding targets". These are functions that cannot be bound to menus in a reasonable way. Among this is "Paste interactive" which does what you look for. Just bind it to a key (Cadence-like is "H"). Also try "Move Interactive" (M) or "Duplicate interactive" (C)

regarding 3: the modifier keys are all taken already. It was suggested once to modify the selection behavior depending on how you drag the rectangle.

regarding 4: I think that means for click selection - don't select cells that are covering the view entirely.

Matthias

olisnr, 2024-11-26:
hallo @klayoutmatthias 
its exactly from this discussion :) 
regarding 1: im perfect happy with the move after selection. my problem, is, that if i forget to press "esc" or press an other key per accident after it, i destroy things when i try to select via rectangle.
if i could have a move-mode, that isnt able to select things, or a move-mode, that jumps back into the select mode after ending the movement, i would feel safe :)
2: thanks, thats what i wanted.
3: yes, the way how to draw is a good way to select the select mode.
4: that was You suggestion, i think. and it sound very reasonable to me.

klayoutmatthias, 2024-11-30:
I have implemented two enhancements and they will go into 0.29.9:

* The status bar now shows what is selected - previously it did not fall back to "nothing". Maybe that helps to remember that you are dragging something for 1. 
* Point-and-click now does not select objects (shapes, instances), if their edges are outside the view. So potentially "hollow" objects can't be selected without noticing.

3. will come later.

Matthias

adrienluitot, 2024-12-01:
Maybe for the selection rectangle (3) it might be an option in setup?

I think most people whether prefer inner selection or touching selection. But you get use to it and adapt the way you select depending on your habit.
I think it's pretty rare to want a inner select and then few minutes after a touching select. Maybe I'm wrong?

This option would be pretty easy to implement and would not require any key modifier or unusual behaviour.

klayoutmatthias, 2024-12-02:
Thanks for the hints.

I can reproduce the issues.

Problem with the key bindings is internal - actually there is are bindings for the same thing (tool bar button and menu entry) and priority changed. That is a old topic I have not taken care of that so far. It needs fixing.

Also, the partial mode object snapping has changed: it is working, but grid snapping has higher priority now. So if the objects you snap to are off-grid, the edges won't snap to the objects. I can try to fix that as well.

Sorry for the inconvenience. It is a very busy time currently and there is too little time to do things properly.

Matthias

klayoutmatthias, 2024-12-02:
I fixed the problem with the partial mode snap and the key binding issue.

Partial mode now does a smarter snapping that both considers the edge snap against plus the grid - the grid only along the edge, not perpendicular to it.

The key binding problem is actually a bug in the old versions, the new one only triggered it.

The next version will properly read the configurations again. However, there is a caveat: if you switch back to old versions, the configuration will already be spoiled for them. So they do not properly read key bindings that go to toolbar items. Even "Reset key bindings "will not help.

In order to fix the configuration back to something useful for old versions, you have to edit the configuration file (in `~/.klayout/klayoutrc` or `c:\user\<you>\KLayout\klayoutrc`): delete the "key-bindings" element and re-configure the key bindings or from the list inside that element delete all entries pointing to "@toolbar..." paths.

klayoutmatthias, 2024-12-02:
This is the PR that should fix the last two problems: https://github.com/KLayout/klayout/pull/1943

olisnr, 2024-12-03:
super! thanks!

a observation with snapping of paths, via the `Paritial` mode was, that sometime i end up off-grid. and i think, its the case if i "grab" the middle-line of the path. i think, if i grab the endpoint, thats never happened.
with the endpoint at the other hand, sometime the path ignores the manhatten-roule in the `Paritial` mode.
but this is from older versions the .29.9...

klayoutmatthias, 2024-12-03:
Yes, well. I played a lot to make the Partial snapping intuitive under object snapping, grid snapping and move constraints. That is really a challenge. With the latest change I thought I got it right - observing grid **and** object snapping, but then grid snapping got too strong.

I am confident the hotfix will provide a better solution. But I will test it first before releasing.

Master is up to date already and there are (unsigned) builds here if you like to test it: https://www.klayout.org/downloads/master/

Matthias

klayoutmatthias, 2025-04-26:
I guess the problem is solved now.

Please open a new ticket if you feel that is not the case.

closed 2025-04-26 by klayoutmatthias