KLayout/klayout #1935: Deep LVS failing

stafverhaegen-chipflow, 2024-11-27:
I have a teststructure that fails `deep` LVS but goes through `flat` LVS. This both for the LVS deck provided by IHP as for my Chips4Makers generated LVS deck. We both use `extend("*")` to get the full design (see also IHP-GmbH/IHP-Open-PDK#254).
I now made an extra test case where I used the global net connection as described in #1813: [testcase_deep2.tar.gz](https://github.com/user-attachments/files/17930764/testcase_deep2.tar.gz). This also fails.
Problem seems to be that the nmoses are extracted at the wrong level in the hierarchy.
I am using klayout v0.29.8

klayoutmatthias, 2024-11-28:
Ah, these lovely SRAM layouts ...

The problem is a known one and the (current) solution are "cheats": https://www.klayout.de/doc-qt5/about/drc_ref_global.html#k_17

Here is for example a discussion with a similar problem: https://www.klayout.de/forum/discussion/comment/11290#Comment_11290

In your case, the fix is:

```
# somewhere at the beginning of the LVS script:
sram_cells = [ "SP6TCCell" ]

...

# upon device extraction add the "cheat" bracket:
cheat(sram_cells) {

  # Transistors
  # sg13g2_hv_nmos
  extract_devices(mos4("sg13_hv_nmos"), {
      "SD" => Activ__conn, "G" => gate__mosfet__sg13g2_hv_nmos, "tG" => GatPoly__conn, "W" => substrate__IHPSG13G2,
  })
  # sg13g2_hv_pmos
  extract_devices(mos4("sg13_hv_pmos"), {
      "SD" => Activ__conn, "G" => gate__mosfet__sg13g2_hv_pmos, "tG" => GatPoly__conn, "W" => NWell,
  })
  # sg13g2_lv_nmos
  extract_devices(mos4("sg13_lv_nmos"), {
      "SD" => Activ__conn, "G" => gate__mosfet__sg13g2_lv_nmos, "tG" => GatPoly__conn, "W" => substrate__IHPSG13G2,
  })
  # sg13g2_lv_pmos
  extract_devices(mos4("sg13_lv_pmos"), {
      "SD" => Activ__conn, "G" => gate__mosfet__sg13g2_lv_pmos, "tG" => GatPoly__conn, "W" => NWell,
  })

}
```

With this modification, LVS passes in my case.

This problem happens frequently with cells sharing devices - specifically, when the drain and source areas of the devices stretch over multiple cells. In your case, this region is shared between the cells:

[image]

KLayout collects all shapes that form source or drain in order to properly derive the size of the device. But the full shape of source or drain only appears one level up in the hierarchy. Here, the difference is only in AD or AS, but basically a device could be split along the gate and extend over two cells. In that case, the width can only be measured in the full device which is not available in the subcell. As a consequence, devices appear one level up.

The "cheat" cheat pretends that for the listed cells there is no neighborhood to consider. So device extraction does not try to collect shapes from adjacent cells and the devices are formed in the subcell.

I agree that is not really nice, but I think, this usually applies to a limited subset of cells (SRAM, decoders, ...).

I have tried an improved implementation that compromises proper extraction of AD/AS in favor a better hierarchical extraction of the devices (branch "device-extraction-improvement"). I will use this problem as a test for this implementation. But I have not released that yet.

Best regards,

Matthias

stafverhaegen-chipflow, 2025-02-11:
@klayoutmatthias 
I'm now looking at this again. Thing is that I want to keep the layout of the single cell DRC clean and that can't be done without the overlapping active.
I understand how the AS/AD extraction promotes the devices to the higher level.
In my design I do have a prBoundary that actually indicates the placement/abutment boundary of the cells. Maybe that layer could be used to extract the as/ad for the cell only inside this layer ? It will then likely be wrong for the outer cells which don't have a neighbour but I think that is acceptable.

stafverhaegen-chipflow, 2025-02-11:
Another comment is that for LVS you likely don't match AS/AD anyway between schematic and layout so maybe a mode where this is taken into account can still keep the devices on the lowest cell ?

klayoutmatthias, 2025-02-12:
Hi Staf,

AS/AD are not taken into account for LVS match, so my plan is to give them second priority or have a switch that allows turning off AS/AD extraction entirely.

I understand that for simulation, having AS/AD may be useful. But I'd count that as a feature for the parasitic extraction topic. 

Allow me a few days to look into that topic.

Best regards,

Matthias

klayoutmatthias, 2025-10-20:
@stafverhaegen-chipflow Thanks for the test case. I will try it on my device extraction improvement branch. However as mentioned, the implementation will compromise the AD/AS extraction and it needs to be switchable. I don't know how to exactly extract AD/AS for the hierarchically shared case.

klayoutmatthias, 2025-10-20:
My old "device-extraction-improvement" branch does not help here, as the problem already happens at computation time of the derived layers :(

So as of now, the `cheat` hack is still all I can offer.

Matthias