KLayout/klayout #1936: connectivity filter for separation check

tvt173, 2024-11-27:
hi @klayoutmatthias 

In Calibre, we can add a connectivity filter to a separation check, so that, for example, we only check the separation between polygons which are not "connected". let's say one layer is a device marker layer and the other is a metal layer. is it possible to do the same in klayout? 

I want to write a separation check between a device marker layer and a metal layer, that only flags violations between shapes which are not connected.

klayoutmatthias, 2024-11-28:
Hi Troy,

Such a thing is available. The idea is the following: some operations, including the booleans and the DRC checks, support "props_eq" and "props_ne" attributes (see https://www.klayout.de/doc-qt5/about/drc_ref_global.html#props_eq).

This means, shapes are checked or used only if their properties are identical (with "props_eq") or not (with "props_ne").

The whole thing is a part of a broader concept that introduces properties in DRC layers (to some extent). See for example here: https://www.klayout.de/doc-qt5/about/drc_ref_layer.html#select_props.

Anyway, as you can extract layers with net information attached as properties, you can use "prop_eq" and "prop_ne" to limit checking to shapes from the same net or different net.

You can find more information and examples here: https://www.klayout.de/doc-qt5/manual/drc_runsets.html (In section "DRC and User Properties").

Matthias

tvt173, 2024-11-28:
thanks! i saw the property filter option and suspected that might be the way to go, but now i also see the relevant examples to help me do what i need. will give it a try

klayoutmatthias, 2024-11-28:
Very good! Let me know if it works the way you need it.

Matthias

tvt173, 2024-11-29:
thanks Matthias, i was able to do what i needed :)