KLayout/klayout #1938: kräsch ät pasting...

olisnr, 2024-12-01:
after hours of working a crash happened at pasting or may be an other key-stroke (some times i miss the correct buttons and dont know what i really did).

[klayout_crash.log](https://github.com/user-attachments/files/17967443/klayout_crash.log)

klayoutmatthias, 2024-12-01:
Serious bug reports please. How should I know what you did? The crash log tells a few interesting things, but not, why it crashed.

Plus try the latest version before you submit a report.

And I don't like that accusation implied. It's not my fault if you forget to save regularly.

Matthias

olisnr, 2024-12-02:
the report was serious meant: i did many things including many `cntl-z` and `y`. what i wanted to do at the time of the crash was pasting a cell. but im not sure if i really pressed `cntr-v` or something different by mistake. 

i'm sorry if it sounded like an accusation. it wasn't that. I was in a good mood and thought that the message might be helpful. even if I don't know what exactly happened.

frequent saving is great. I know that I am responsible for this, and I always do this when I think that I don't want to lose the work I have done. that's why I always change `ctrl+s` to `file_menue.save` as well.

i will try to seem more serious in the future!

sebastian-goeldi, 2024-12-02:
There was a bug with 0.29.7 that might be related to copy/paste. So if you are on that version, please upgrade

https://github.com/KLayout/klayout/issues/1874

olisnr, 2024-12-02:
thanks. i download the newest version!

closed 2024-12-02 by olisnr