KLayout/klayout #1941: Version 0.29.9 crash

Kazzz-S, 2024-12-02:
Hello @klayoutmatthias,

Thank you for making the maintenance release of 0.29.9.
Unfortunately, I'm facing a crash when opening a simple design. [gds.zip](https://github.com/user-attachments/files/17971564/gds.zip)

----

[image]
[image]

----

**Valgrind on Linux**
[image]
[KL0299-Mint203-Valgrind.zip](https://github.com/user-attachments/files/17971563/KL0299-Mint203-Valgrind.zip)

----

I'm also currently building it on my Mac. I could add more information tonight.

Best regards,
Kazzz-S

klayoutmatthias, 2024-12-02:
Hi Kazzz-S.You are right. I have no idea why the released version does that. I was working nicely for my builds.I will check what is wrong. I think I will create a hot fix for that problem.MatthiasSent from Android device

KL0299-Win11.png (view on web)
KL0299-Mint203.png (view on web)

KL0299-Mint203-Valgrind.png (view on web)

Kazzz-S, 2024-12-02:
Yes!

[image]

Kazzz-S

klayoutmatthias, 2024-12-02:
I have fixed the problem. 

As 0.29.9 is not quite useful with the bug, it makes sense to wait for 0.29.10 before creating the DMGs.

Sorry the for inconvenience. It's very busy right now. Too little time for testing :(

Matthias

closed 2024-12-03 by klayoutmatthias

Kazzz-S, 2024-12-03:
[image]

Kazzz-S

klayoutmatthias, 2024-12-04:
Thanks a lot for testing!

I just released 0.29.10. I hope this is a good one then :)

Best regards,

Matthias