KLayout/klayout #1942: PythonStandalone Package 0.29.9 stubs errors

sebastian-goeldi, 2024-12-02:
Hi Matthias,

I think my request for the `Point.move(d)` confused your autogeneration  script for the stubs :face_with_peeking_eye: 

In the `dbcode.pyi` at line 704 there is a syntax error:

```python
    @overload
    def moved(self, dx, 0: int, dy: Optional[int] = ...) -> Box:
        r"""
        @brief Moves the box by a certain distance

        This is a convenience method which takes two values instead of a Point object.
        This method has been introduced in version 0.23.

        @return The moved box.
        """
        ...
```

The `dx, 0: int,`  should be `dx: int = 0` I think.

Hopefully this is just some minor tweak to fix it.

If you would want help with this, as always, I'd be happy to help.

Best,
Sebastian

klayoutmatthias, 2024-12-02:
Ah, well that was a stupid typo that was not covered by tests :(

But as a bonus I added defaults for "Box#enlarge" and "Box#enlarged" which makes a lot of sense to me:

```
box.enlarge(dx = 17)
```

closed 2024-12-03 by klayoutmatthias

klayoutmatthias, 2024-12-03:
Sorry, did not want to close this issue. It happened automatically when I merged the PR.

One question: is there an easy way to test .pyi files? I mean in a sense of syntax correctness.

Matthias

klayoutmatthias, 2024-12-03:
Oh, i guess I found it: 

```
python3 -m mypy.stubtest klayout.dbcore
```

If does not give syntax errors anymore, but a lot of warnings - mainly because it does not understand how the overload inference works inside KLayout ...

Matthias

sebastian-goeldi, 2024-12-03:
I think so. You can install the package with python and install mypy and then runn `mypy -p klayout`

So `python -m pip install klayout mypy; mypy -p klayout` should do the trick.

mypy is a static type checker and will check all the type hints for correctness. Similar to pylance etc

sebastian-goeldi, 2024-12-03:
This works (probably also works with stubtests):

`mypy -p klayout --disable-error-code overload-cannot-match --disable-error-code no-untyped-def --disable-error-code no-untyped-call --disable-error-code misc --disable-error-code attr-defined --disable-error-code unused-ignore --disable-error-code override --disable-error-code name-defined --disable-error-code no-redef --disable-error-code import-not-found --disable-error-code assignment`

Might be worth to fix some of the errors (like unused ignore redef etc, or at least to have a look at them and then use ignore commands selectively)