KLayout/klayout #1953: [PcellDeclarationHelper] callback_impl & coerce_parameters_impl parameter display issue

dimchrist89, 2024-12-10:
Hi, Matthias.

I am using klayout v.0.29.8 and I am seeing issues with visibility parameter display. I have implemented a dummy pcell code, which triggers this behavior:

`import pya

class Foo(pya.PCellDeclarationHelper):

    def __init__(self):

        super(pya.PCellDeclarationHelper, self).__init__()

        self.param( 'ListParam',
            self.TypeList,
            "ListParam",
            choices = [['Choice1', 'Choice1'], ['Choice2', 'Choice2']],
            default = 'Choice1'
        )

        self.param( 'StringParam',
            self.TypeString,
            "StringParam",
            default = 'Dummy'
        )

    def coerce_parameters_impl(self):
        if self.ListParam == 'Choice2':
            self.ListParam = 'Choice1'

    def callback_impl(self, name):

        self.StringParam.visible = self.ListParam.value == 'Choice1'

    def produce_impl(self) -> None:

        text_obj  = pya.DText('FOO', 0, 0)
        self.cell.shapes(0).insert(text_obj)`

This issue here is that when i select 'Choice2' as a value for 'ListParam', then this value properly reverts to 'Choice1' (based on coerce_parameters_impl implementation), but 'StringParam' remains hidden, which based on callback_impl should only happen with value 'Choice2'. This may be alleviated by checking/unchecking 'Show Parameter Names' button (which I assume that triggers some state refresh).

klayoutmatthias, 2024-12-11:
Thanks, I will look into that.

I guess it is a problem of execution order - callbacks before coerce_parameters.

I wonder however, if it is good idea to change that. Basically, callbacks are first tier, so they act immediately when you trigger some change or push a button. In your specific case, this means that you could assume that coerce_parameters has not been called yet. 

Matthias

dimchrist89, 2024-12-11:
Apologies for spamming this thread, but I believe it is relevant to the previous issue and I believe something may be glitchy in how the parameter states are stored. 

I extended the previous example and used only callback_impl for both parameter value & display updates. 

`import pya

class Foo(pya.PCellDeclarationHelper):

    def __init__(self):

        super(pya.PCellDeclarationHelper, self).__init__()

        self.param( 'ListParam',
            self.TypeList,
            "ListParam",
            choices = [['Choice1', 'Choice1'], ['Choice2', 'Choice2']],
            default = 'Choice1'
        )

        self.param( 'ListParam1',
            self.TypeList,
            "ListParam1",
            choices = [['Choice1', 'Choice1'], ['Choice2', 'Choice2']],
            default = 'Choice1'
        )

        self.param( 'StringParam',
            self.TypeString,
            "StringParam",
            default = 'Dummy'
        )

        self.ListParamValue = 'Choice1'

    def callback_impl(self, name):

        if name:
            if self.ListParam.value == 'Choice2': 
                self.ListParam1.value='Choice1'
            else:
                self.ListParam1.value='Choice2'
        else:
           self.ListParam1.value='Choice2'

        self.StringParam.visible = self.ListParam1.value == 'Choice1'

    def produce_impl(self) -> None:
        pya.Logger.info(self.ListParamValue)

        text_obj  = pya.DText(self.ListParamValue, 0, 0)
        self.cell.shapes(0).insert(text_obj)`

When I access the instance parameters, via Instance.pcell_parameters_by_name, I get inaccurate results

[image]

It seems that doing multiple callback_impl calls (combination of empty  and non empty 'name' parameter AND taking this into account within a conditional) triggers the issue.

klayoutmatthias, 2024-12-13:
No worries, no spamming seen :)

Thanks for the second sample. Honestly what you are doing feels a bit odd recalling the use cases I had in mind for the callbacks. But I can't pinpoint a particular fault here, neither on your side nor on KLayout side. Still, there may be a plain bug.

But let me explain the intention of callbacks: those are simply called (as the name says) when some value is changed. That is supposed to implement some hide/disable logic like you do. Changing values is a bit outside that scope. The reason is that if you use the value-changing features and expect a callback, it would be very easy to create a infinite loop where a callback changes a value back and forth for example. Hence, changing a value inside a callback should not trigger another callback. This is not a problem here, but it illustrates a perception problem: a callback is not a tool to *ensure* coherence between values. Instead, it is mainly meant for showing or hiding parameters for example when a specific option is selected.

On the other hand, "coerce_parameters" is a global validation function meant to ensuring parameter validity under conditions beyond user interaction. This function is called frequently to prevent the PCell code from receiving invalid requests - like billion finger devices or meter-sized via arrays - that would crash the tool. But this function cannot change the parameter display states.

I feel there is need for more guidance and I need to wrap my head around your use case. And still of course there might be a bug too.

Matthias

dimchrist89, 2024-12-13:
Hi again matthias. 

Usually, in EDA platforms callbacks also do change the values of the parameters, ( i have seen multiple pdks and all use the same approach, i guess that is up to the developer to ensure non-recursiveness), but i understand the different approach. The issue when using both coerce and callback in my case comes from the fact that these are called multiple times. For example, by using only callback_impl and by changing a value, i saw the following execution order: callback(name)->produce->callback('')->callback(''), which means that the two final callback calls may alter the pset, thus creating inconsistencies between the layout and the parameters. 

 If i use coerce, then it needs to be ensured that when within callback_impl, the states are properly refreshed, which seems logical to me, since display conditions are usually dependent of the parameter set (other pvalues, etc). Unfortunately, In the latter case, I honestly felt a little bit overwhelmed by the execution order, so i dont have a concrete sample case yet

klayoutmatthias, 2024-12-13:
Yes, I know this behavior of PDK PCells. And I hate it, because once some "smart logic" changed your value to something, it's hard to switch back to the original value. You can do a Cancel and start over. That is why I originally favored a hidden "coerce_parameters" function. You can raise exceptions there, which are displayed and you can thus indicate some out-of-bounds conditions. But it will not change what you entered.

Virtuoso is a great tool (I am not paid by Cadence), but it favors some bad habits. And I am not speaking of marketing a scripting language from the age of vacuum tube computers.

```callback('')``` is a global refresh. Callbacks are tied to Qt events and can happen under different conditions. There is no warranty about the order of execution.

I see a consistent behavior in your case if I use this code:

```
class Foo(pya.PCellDeclarationHelper):

  def __init__(self):
  
      super(pya.PCellDeclarationHelper, self).__init__()
  
      self.param( 'ListParam',
          self.TypeList,
          "ListParam",
          choices = [['Choice1', 'Choice1'], ['Choice2', 'Choice2']],
          default = 'Choice1'
      )
  
      self.param( 'ListParam1',
          self.TypeList,
          "ListParam1",
          choices = [['Choice1', 'Choice1'], ['Choice2', 'Choice2']],
          default = 'Choice1'
      )
  
      self.param( 'StringParam',
          self.TypeString,
          "StringParam",
          default = 'Dummy'
      )
  
  def coerce_parameters_impl(self):
      if self.ListParam == 'Choice2': 
          self.ListParam1 = 'Choice1'
      else:
          self.ListParam1 = 'Choice2'
  
  def callback_impl(self, name):
      self.StringParam.visible = self.ListParam.value == 'Choice2'
  
  def produce_impl(self) -> None:
      text_obj  = pya.DText(self.ListParam1, 0, 0)
      self.cell.shapes(self.layout.layer(1, 0)).insert(text_obj)
      self.cell.shapes(self.layout.layer(1, 0)).insert(pya.DBox(-0.1, -0.1, 0.1, 0.1))
 ```

Note changes to "produce_impl": I do not use the layer index 0 directly and that may not be defined (yet) and added a box to improve visibility. I also don't use "ListParam1Value". This is a member of the PCell class, but the state is undefined as it is not maintained by KLayout. Simply use "ListParam1" to access the value - except for the callback, where "ListParam1" is a struct and you need to use "ListParam1.value".

The code works because it is independent from the order `coerce_parameters` and `callback` happens. The callback does not need the outputs from `coerce_parameters` as it takes the primary information. 

I am aware that this may not be applicable in your actual case, but basically it should be possible to implement value-changing logic inside "coerce_parameters" and state-changing logic inside "callback", when you base the latter on your primary inputs.

Yet I see some inconsistencies - for example, changing the value of the choice in the callback is not reflected in the drop-down box (or not immediately). So there is at least this bug.

Thanks,

Matthias

dimchrist89, 2024-12-13:
I tried the approach you are describing: Modify values in 'coerce', modify visibility in 'callback'. 
I saw the following execution order callback(name) -> coerce -> coerce -> coerce - produce -> callback('') -> callback('')
So, if i were to update the values and then use the values to evaluate visibility, I cannot see how to do it with the flow above. 
Am I missing something here?

dimchrist89, 2024-12-14:
Another thing to note is that, updating the parameter values may dependent on the parameter that initiated the update. For example, assume I have an arc with an inner and outer radius parameters, with constant width. If i want to modify the inner radius, then the outer must be refreshed, while if i want to do the opposite, the inner must be the one that refreshes (there must be some "prioritization", that having the "name" parameter allows for, and coerce is not aware of. I tried a work around (saving the "name" in an instance property and then reading it within coerce), but given that the execution order is what it is, i feel no confident to use it

klayoutmatthias, 2024-12-14:
Hi @dimchrist89,

The execution order of the various callback and coerce_parameters calls is not defined. Callbacks originate from the Qt event system. Their order and frequency depend on the OS or Window Manager. "coerce_parameters" is called internally and in various occasions. There is caching involved, so "coerce_parameters" may be skipped, as can be "produce".

If you have mutually dependent parameters, you cannot know which parameter has changed from the callbacks. There is also the possibility of programmatic instantiation of the PCell - in that case, no callbacks are involved at all. Another thing I hate about some PDK PCells BTW. I wish PCell designers would care more for layout automation requirements ...

There are several strategies for dependent parameter implementation under these constraints.

Solution 1: let the user decide. Implement a choice parameter that decides which is the primary and which is the secondary parameter. For example, a capacitor PCell my offer to specify Area or Capacitance. There are three parameters: mode (Area or Capacitance), area and capcitance. In area mode, the capacitance is read-only and the value is computed from the area. In capacitance mode, the area is read-only and computed from the capacitance. The state of the parameters depends on the mode parameter and is set inside "callback". The values are computed inside "coerce_parameters". 

Solution 2: use hidden parameters to track the state. This concept is demonstrated by the built-in Circle sample PCell. This PCell has two ways of configuring the radius: by a numerical entry box and by moving the handle. There is a hidden "effective radius" parameter that allows to decide which parameter was changed inside "coerce_parameters". The distance of the handle to the circle center is used to compute the graphically set radius. If different from the effective radius, both effective and entry box value are updated. In the other case, the entry box value will be used to update the effective radius and the handle position. "produce" uses the effective radius only.

In both cases, circular dependencies are avoided and information about the source of a change is not required.

Matthias

dimchrist89, 2024-12-16:
> Yet I see some inconsistencies - for example, changing the value of the choice in the callback is not reflected in the drop-down box (or not immediately). So there is at least this bug.
> 
> Thanks,
> 
> Matthias

Hi, Matthias. Will this bug be planned for a future release? Would you like me to file a new ticket?

klayoutmatthias, 2024-12-16:
After reviewing the problem, I think that was because of this line in the original code in the unconditional branch of the callback:

```
       self.ListParam1.value='Choice2'
```

As you observed, this branch got called on other occasions too, hence resetting the value.

To be honest, I do not see a particular issue as of now. Maybe you want to complain about `callback('')` being called too often.

dimchrist89, 2024-12-16:
> Hi, Matthias.
>
> I am using klayout v.0.29.8 and I am seeing issues with visibility parameter display. I have implemented a dummy pcell code, which triggers this behavior:
>
> `import pya
>
> class [...]

I followed your guidance and used the coerce_parameters for value update and callback_impl for visibility state. Going back to my initial post, I still see this issue. Do you consider this an issue that should be solved? If not, should I follow a different approach to get it working? Thank you

dimchrist89, 2024-12-16:
`import pya

class Foo(pya.PCellDeclarationHelper):

    def __init__(self):

        super(pya.PCellDeclarationHelper, self).__init__()

        self.param( 'ListParam',
            self.TypeList,
            "ListParam",
        choices = [['Choice1', 'Choice1'], ['Choice2', 'Choice2']],
        default = 'Choice1'
         )

        self.param( 'StringParam',
             self.TypeString,
            "StringParam",
            default = 'Dummy'
         )

    def coerce_parameters_impl(self):

        if self.ListParam == 'Choice2':
            self.ListParam = 'Choice1'

        pya.Logger.info(f'Inside coerce_parameters_impl, ListParam value:  {self.ListParam}')

    def callback_impl(self, name):

        self.StringParam.visible = self.ListParam.value == 'Choice1'

        pya.Logger.info(f'Inside callback_impl, ListParam value: {self.ListParam.value}, StringParam visibility: {self.StringParam.visible}')

    def produce_impl(self) -> None:

        text_obj  = pya.DText('FOO', 0, 0)
        self.cell.shapes(0).insert(text_obj)
`

just added some print statements and while callback_impl is called twice and StringParam.visible is True, the parameter is invisible in the form, thus the issue is not the execution order per se. ps I cleared the log after the form, then I tried to change the ListParam value from choice1 to choice2, so the print messages here do not come from previous steps

[image]

klayoutmatthias, 2024-12-18:
Is that Windows or Linux?

dimchrist89, 2024-12-18:
It is Linux (sorry, I was not aware that both are supported)

klayoutmatthias, 2024-12-18:
Okay, thanks for the screenshots.

Let me summarize: the expectation is that the final two calls of "callback" should set the visibility to "True". That happens, but the visibility does not change. So that is a bug, you're right.

I will debug the problem. Thanks for reporting.

Matthias

klayoutmatthias, 2024-12-18:
I have published a patch in the attached PR. 

To me the behavior looks consistent now - switching to "Choice2" will immediate snap back to "Choice1" and the string parameter stays enabled.

Do you have a way to test the branch with your specific requirements?

Best regards,

Matthias

dimchrist89, 2024-12-19:
Hi Matthias,

I verified your fix. Thank you for the timely response!

klayoutmatthias, 2024-12-19:
You're welcome :) I hope the discussion was helpful as well. 

I will release the fix with the next minor release and thanks for your patience bringing the issue to the right point.

Best regards,

Matthias

dimchrist89, 2024-12-19:
Just some suggestions regarding the current topic

1) Some extra documentation would be really helpful regarding how the pcelldeclararionhelper function is supposed to be used, especially when using the newly added callback_impl functionality. I understand the effort you put into this project, so if you need contributors, i would be happy to help😄

2) Maybe some pcellparameterstate properties should only be readonly, like value, since as far as i understood, callback_impl is not supposed to be used for that kind of manipulation (??)

Cheers!

closed 2025-01-04 by klayoutmatthias