KLayout/klayout #1954: [Advocacy]Can all of use contribute to a PCell tutorials of KLayout？

luomeng007, 2024-12-13:
Guys, I am doing a job related to plot chip layout. And I tried a lot of softwares, like IPKISS, GDSFactory and so on. However, I find all of them output result into KLayout. Thus, I want to ask , why not all of us focus to KLayout? Why not to write a tutorial for PCell crating directly by using KLayout? If we do so, we do not need api any more!!!

sebastian-goeldi, 2024-12-13:
There already is https://www.klayout.de/doc/programming/ruby_pcells.html .

That being said, if I remember correctly, the klayout PCells are not really intended to be used for assembling entire chips. Reason being, they need to serialize the input parameters in order to make them accessible through the GUI and also play nicely with it. So while it is possible, see how much "overhead" you need to create in order for a KLayout PCell to be created.

If you go the standard python API (i.e. not through the PCells but rather manipulate cells directly), then you probably still want some additional functionalities. And that's essentially what gdsfactory is nowadays. It is a nice and opinionated dress-up around klayout (ok that might be slightly unfair as it provides integrations and so on for other tools as well) mainly targeted for photonics (so you will have a bad time if you want to  assemble a digital design with it).

You don't need no API if you don't want, you can always take away a package on top if you don't like it, until you arrive at writing your own GDSII writer ofc.

And I wouldn't call it entirely fair to say that the tools output to klayout. It's often used as the viewer, one big reason being so stable and also one of the only few options for free (& open source) viewers.

klayoutmatthias, 2024-12-13:
Although I feel honored by the proposal, I think having alternatives is necessary in a healthy ecosystem. Hence I appreciate GDSFactory providing a different approach, and there is a close collaboration which I wish to continue :)

I see KLayout as a graphical tool for macro layout, while GDSFactory implements a programmatic approach on a higher, above-polygon level. In the end it's up to you to pick the solution that is best for you - as long as there is interoperability and parts can be integrated into a final product. I think we're on a good track there.

Matthias

luomeng007, 2024-12-17:
Ok， thanks for you expnation. Actually I think GDSFactory is a great software, you know, I also use IPKISS, the IPKISS uses python class to define PCell while GDSFactory use decorators, in my mind, maybe class is better for inherit? After all, Python is a kind of language that face to object. How do you feel?

sebastian-goeldi, 2024-12-17:
The reason gdsfactory went the function route instead of classes is the one you outlined below: Few people like programming and even fewer understand OOP. Gets even more complicated if you bring inheritance to the table.

In theory, you are free to use a class based approach with gdsfactory, you just have to do the caching yourself (shouldn'tactually be too hard).

Wrt the GUI approach, that's exactly (partially) what the (somewhat new) company around gdsfactory is doing. The problem is, it takes an incredible amount of work to make a frontend and make it reliable (if you e.g. look at the issues in klayout for fixing new or hidden bugs related to Qt or weird interactions with it). We think it's not viable (in a reasonably short timeframe) to make it in the open as FOSS without sacrificing even more of our free time. I honestly don't know how Matthias managed to create such an amazing tool both as a GUI and scripting package, but my guess would be hard work over many years and sacrificing tons of free time.

luomeng007, 2024-12-17:
Hhh, I see. You are right, it will definitely occupy our free time. But that is a great job. Once we realize that, we benefit all mankind. The world needs crazy people.

My idea is that maybe you could just let more people know the job you are doing and assign some task or enhancement items to prople who are interested in this area. The more people join the community, the quicker the ultimate goal we could realize. Before I tried to look through the tutorial and docs of GDSFactory, I think it is not in detail enough for people to join in. For my side, I create some tutorials by myself which I feel much better for people to understand the usage of GDSFactory.  In a word, my suggestion is that send 'bricks or blocks' to users and let them build their own skyscrapers, and then these skyscrapers can be shared in GDSFactory and make other people quickly get involved. Like a collection of forum and databese(like a website or something?). The 'bricks or blocks' here I mean not the PCells, they are some routines and ideas behind the GDSFactory, then people who want to join can get your idea and discuss efficiently with you. I feel by doing so, our community can grow up quickly and attract more brilliant guys. Then you find the balance of free time and development of community and we also can accelerate the progress to realize the ultimate goal.

sebastian-goeldi, 2024-12-17:
Gdsfactory would be more than happy to get your own tutorial for an alternate approach. The problem of the tutorial/notebooks is always to strike a balance. Some people want to build their own from bricks and mortar like you, but more often than not, people just want to snap their finger 5 times and get 4x walls + a roof.

Also, it's often times very hard to write appropriate docs if the subject is somewhat complex. Often you will skip the steps that are very clear for you, but cause lots of people to struggle on their own. Help with improving docs (especially in this sense) is always very much appreciated.

closed 2024-12-17 by luomeng007