KLayout/klayout #1961: [bug] Python Script

luomeng007, 2024-12-24:
I want to write an export data program. And if I wrap the following code into a function, I get correct result. But when I wrap them into a class structure. I get an error.
```
import pya

class Visualize:
    def visualize01(obj,
                    cell_name: str = 'Die',
                    layer: int = 1,
                    datatype: int = 0):
        layout = pya.Layout()
        cell = layout.create_cell(cell_name)
        layer = layout.layer(pya.LayerInfo.new(layer, datatype))
        cell.shapes(layer).insert(obj)
        layout.write(f'./Layouts/{cell_name}.gds')

if __name__ == '__main__':
    obj_ = pya.DPolygon.new([pya.DPoint(0.0, 0.0),
                             pya.DPoint(1000.0, 0.0),
                             pya.DPoint(1000.0, 1000.0),
                             pya.DPoint(0.0, 1000.0), ])
    view = Visualize()
    view.visualize01(obj_)
```
I get an error like below.

> Traceback (most recent call last):
  File "C:\Project\PythonProject\KLayout\Tutorials\Visualizer\Visualizes.py", line 31, in <module>
    view.visualize01(obj_)
  File "C:\Project\PythonProject\KLayout\Tutorials\Visualizer\Visualizes.py", line 21, in visualize01
    cell.shapes(layer).insert(obj)
TypeError: No overload with matching arguments. Variants are:
  void insert(const Shapes shapes, unsigned int flags) [no value given for argument #2]
  void insert(const Region region, const ICplxTrans trans) [no value given for argument #2]
  void insert(const Region region, const DCplxTrans trans) [no value given for argument #2]
  Shape insert(const Shape shape, const ICplxTrans trans) [no value given for argument #2]
  Shape insert(const Shape shape, const Trans trans) [no value given for argument #2]
  Shape insert(const Shape shape, const DTrans trans) [no value given for argument #2]
  Shape insert(const DBox box) [match candidate]
  Shape {insert|insert_path}(const Path path) [match candidate]
  Shape {insert|insert_box}(const Box box) [match candidate]
  Shape insert(const Shape shape, const DCplxTrans trans) [no value given for argument #2]
  void insert(const RecursiveShapeIterator iter, const ICplxTrans trans) [no value given for argument #2]
  void insert(const Shapes shapes, const ICplxTrans trans) [no value given for argument #2]
  void insert(const Shapes shapes) [match candidate]
  void insert(const Edges edges) [match candidate]
  void insert(const EdgePairs edge_pairs) [match candidate]
  void insert(const Region region) [match candidate]
  Shape insert(const Shape shape) [match candidate]
  void insert(const RecursiveShapeIterator iter) [match candidate]
  void insert(const Texts texts) [match candidate]
  Shape {insert|insert_point}(const Point point) [match candidate]
  Shape insert(const DText text) [match candidate]
  Shape {insert|insert_simple_polygon}(const SimplePolygon simple_polygon) [match candidate]
  Shape insert(const DSimplePolygon simple_polygon) [match candidate]
  Shape {insert|insert_polygon}(const Polygon polygon) [match candidate]
  Shape insert(const DPolygon polygon) [match candidate]
  Shape insert(const DPoint point) [match candidate]
  Shape {insert|insert_text}(const Text text) [match candidate]
  Shape insert(const DEdge edge) [match candidate]
  Shape insert(const EdgePair edge_pair) [match candidate]
  Shape insert(const DEdgePair edge_pair) [match candidate]
  Shape insert(const DPath path) [match candidate]
  Shape {insert|insert_edge}(const Edge edge) [match candidate]
  void insert(const EdgePairs edge_pairs, const ICplxTrans trans) [no value given for argument #2]
  Shape insert(const DBox box, unsigned long long property_id) [no value given for argument #2]
  Shape insert(const DEdge edge, unsigned long long property_id) [no value given for argument #2]
  void insert(const Texts texts, const DCplxTrans trans) [no value given for argument #2]
  void insert(const Edges edges, const ICplxTrans trans) [no value given for argument #2]
  Shape {insert|insert_path_with_properties}(const Path path, unsigned long long property_id) [no value given for argument #2]
  Shape insert(const DPath path, unsigned long long property_id) [no value given for argument #2]
  Shape {insert|insert_edge_with_properties}(const Edge edge, unsigned long long property_id) [no value given for argument #2]
  Shape {insert|insert_simple_polygon_with_properties}(const SimplePolygon simple_polygon, unsigned long long property_id) [no value given for argument #2]
  Shape insert(const DPolygon polygon, unsigned long long property_id) [no value given for argument #2]
  void insert(const Texts texts, const ICplxTrans trans) [no value given for argument #2]
  void insert(const EdgePairs edge_pairs, const DCplxTrans trans) [no value given for argument #2]
  void insert(const Edges edges, const DCplxTrans trans) [no value given for argument #2]
  Shape {insert|insert_box_with_properties}(const Box box, unsigned long long property_id) [no value given for argument #2]
  Shape insert(const EdgePair edge_pair, unsigned long long property_id) [no value given for argument #2]
  Shape insert(const DEdgePair edge_pair, unsigned long long property_id) [no value given for argument #2]
  Shape {insert|insert_text_with_properties}(const Text text, unsigned long long property_id) [no value given for argument #2]
  Shape insert(const DText text, unsigned long long property_id) [no value given for argument #2]
  Shape insert(const DSimplePolygon simple_polygon, unsigned long long property_id) [no value given for argument #2]
  Shape {insert|insert_polygon_with_properties}(const Polygon polygon, unsigned long long property_id) [no value given for argument #2]
  void insert(const Shapes shapes, unsigned int flags, const ICplxTrans trans) [no value given for argument #2 and following]
 in Shapes.insert

luomeng007, 2024-12-25:
I have already get answer from forum side. Thanks guys.

closed 2024-12-25 by luomeng007

sebastian-goeldi, 2024-12-27:
@luomeng007 could you maybe link the post here, just for completeness sake? ;)

To avoid this: https://xkcd.com/979/ :D

luomeng007, 2024-12-30:
Fine, here it is. [The anwser](https://www.klayout.de/forum/discussion/2644/can-not-get-rounded-end-by-using-pya-path-command#latest)