KLayout/klayout #1962: klayout build problem (windows, MSVC2017)

xushuyi98, 2024-12-28:
Hi Matthias:
I am trying to build klayout using "build.bat -bits <bits dir>", before that i have done the builds for klayout_bits4msvc2017-4.0, but why it still told me "ERROR: no 3rd party binary kit found in D:\klayout\klayout_bits4msvc2017-4.0\msvc2017\msvc2017\x64"?
[image]
Please teach me how to solve it, many thanks!

xushuyi98, 2024-12-28:
I wrote the bits path like "D:\klayout\klayout_bits4msvc2017-4.0\msvc2017"
why the error shown the path "D:\klayout\klayout_bits4msvc2017-4.0\msvc2017\msvc2017\x64"
it doesn't match right? the path was wrong
I think it should be "D:\klayout\klayout_bits4msvc2017-4.0\msvc2017\x64"
so i tried to delete "msvc2017\x64" and wrote the path="D:\klayout\klayout_bits4msvc2017-4.0"
but anther error appears

klayoutmatthias, 2024-12-29:
```
build.bat -bits d:\klayout\klayout_bits4msvc2017-4.0
```

should work. If not, something is wrong with the bits installation. As I don't have the details, I can't help.