KLayout/klayout #1969: Add generic types to dbcore.pyi

MatthewMckee4, 2025-01-11:
I am unsure if this has every been explored, or if its possible. But in the db "h" and "cc" files, generic types are used to help with the units used in a class (float or int). I think this would be useful to show 

```python
from typing import Generic, TypeVar

_TCoord = TypeVar("_TCoord", int, float)

class BaseBox(Generic[_TCoord]):
    bottom: _TCoord
    etc...

class Box(BaseBox[int]): ...
class DBox(BaseBox[float]): ...
```

Ideally this would provide no functional change, it would just:
- Reduce the duplicate code in the dbcore.pyi file
- Help with type hinting and show that these classes are functionally the same. In kfactory we have classes that wrap and use kfactory classes and this would majorly help with our type hinting and code duplication.

Thanks.

klayoutmatthias, 2025-01-16:
The .pyi files are generated automatically from the declarations and there is no annotation that allows deducing generic base types. In the end, that maybe boils down to writing the .pyi files manually. I don't want to do that.

I personally feel that this request offers too little benefit for a pretty substantial implementation effort.

MatthewMckee4, 2025-01-16:
Thanks for getting back to me, that's very reasonable, I'll close this now as i understand it was a big ask. Thanks

closed 2025-01-16 by MatthewMckee4