KLayout/klayout #1971: Suppress warnings written with `tl::warn` from Python

nikosavola, 2025-01-14:
Hi Matthias, great work on the project.

I was using the Spice reader and have many warnings like
> Warning: Element type 'X' ignored in ...

I would like to suppress these but seems this is not possible from the Python API yet as the underlying `tl::warn` is printed unconditionally.

There appears to be a verbosity level for the logger (https://www.klayout.de/doc/code/class_Logger.html#method17) but this does not affect the warnings.

### Suggestion

Allow setting some verbosity level which suppresses even warnings (calls to `tl::warn`).

klayoutmatthias, 2025-01-16:
Makes sense.

The same problem applies to error messages. My suggestion is to expand the verbosity level below 0, with 0 remaining the default. 

Like -10 to disable warnings and -20 to disable errors too. That follows the step-10 verbosity level philosophy that I intended to use for fine-grained control by sub-levels.

I will try if that is possible.

Matthias

klayoutmatthias, 2025-01-16:
I implemented the proposed scheme, but I noticed another issue while debugging:

```
import klayout.tl as ktl

ktl.Logger.verbosity = -10
```

Does **not** switch the verbosity. The reason is buried deeply inside the Python interpreter - at least with version 3.12: while class property getters are available the usual way, setters are re-routed to the metaclass. I don't want to change the whole system to use metaclasses now as I don't want to risk spoiling the release. Instead, I provide an alias now that sets verbosity correctly:

```
import klayout.tl as ktl

# with KLayout 0.29.11:
ktl.Logger.set_verbosity(-10)
```

With that scheme, warnings can be disabled using "-10" and with "-20" errors are disabled too:

```
import klayout.tl as ktl

for level in [ 0, -10, -20 ]:

  print(f"Setting verbosity level to {level}.")

  ktl.Logger.set_verbosity(level)

  ktl.Logger.error(f"Error at level {ktl.Logger.verbosity}")
  ktl.Logger.warn(f"Warning at level {ktl.Logger.verbosity}")
```

will give:

```
Setting verbosity level to 0.
ERROR: Error at level 0
Warning: Warning at level 0
Setting verbosity level to -10.
ERROR: Error at level -10
Setting verbosity level to -20.
```

Matthias

nikosavola, 2025-01-17:
> I implemented the proposed scheme, but I noticed another issue while debugging:
>
> ```
> import klayout.tl as ktl
>
> ktl.Logger.verbosity = -10
> ```
>
> Does **not** switch the verbosity. The reason is buried [...]

Sounds great, thanks for the effort!

closed 2025-01-17 by klayoutmatthias