KLayout/klayout #1974: pya.Technology.technology_by_name returns different object each time

lukasc-ubc, 2025-01-18:
Hi Matthias,

When calling pya.Technology.technology_by_name, using the same technology name, I would expect to review the same object each time.  Namely if the technology is already registered, the object should be the same one, and I should be able to modify/add to the technology object.  But instead, I see ecah call returns a different object.  How can this be, and is there a way to return the same one?

```
> pya.Technology.technology_by_name('EBeam')
<klayout.pyacore.Technology object at 0x16cbf8250>
> pya.Technology.technology_by_name('EBeam')
<klayout.pyacore.Technology object at 0x16cbf89d0>

```

thank you

klayoutmatthias, 2025-01-18:
Hi Lucas,

that is actually intended. Or I should say: there is no other way to do that in the framework of KLayout.

Let me explain: when an object is create on C++ side (as the Technology object), it does now know that it will be used in Python or Ruby or in a different place. Initially, the object is a pure C++ object. 

When you retrieve the Technology object, KLayout will create a Python object for you. But this Python object does not replace the C++ object. Instead it just acts as a proxy to it. If you retrieve the Technology object a second time, a new Python object will be created acting as another proxy to the same C++ object.

You can see that here:

```
> t1 = pya.Technology.technology_by_name("scmos")
> t1
<klayout.pyacore.Technology object at 0x7b19d2b4b380>
> t2 = pya.Technology.technology_by_name("scmos")
> t2
<klayout.pyacore.Technology object at 0x7b19d2b4b410>
> t1.name = "new_name"
> t1.name
'new_name'
> t2.name
'new_name'
```

So when changing the name of t1, the name of t2 changes as well as both Python objects point to the same C++ object.

(Note: for some classes, these proxies act like smart pointers - i.e. they know when the C++ object is destroyed and become invalid. `Technology` is not of this kind: it is rather a dumb pointer. So when you remove a technology, make sure you do not keep a Python object pointing to that. Accessing this Python object after the Technology is removed may crash the application. Maybe I will change that for improved robustness.)

An exception from the "Python object is a proxy" scheme are objects **created** on Python side. For example, you may create a custom Library in Python: you implement your own Library class and register it. In that case, the object lives in Python space. When you retrieve such an object using `pya.Library.library_by_name(...)`, you will see the same Python object that you have registered and every call of `library_by_name` will render the same object.

If you care about object identity: in case of the `Technology` object, the name is unique and comparing the name will tell you if the same C++ object is addressed.

Matthias

lukasc-ubc, 2025-01-20:
Hi Matthias,

That explains it very well.  Thank you.

What I was trying to do is add additional variables to the Technology, for custom functionality.   Specifically, I wanted to add optical waveguide type definitions to the technology, at run time.  Then I wanted to read them back later.  I will look for a different way.

regards
Lukas

klayoutmatthias, 2025-01-20:
Hi Lukas,

I think I understand. That is a general limitation: C++ objects per se do not have ways of attaching to Python, so you cannot store additional information with the technology object.

For Cell and Layout objects we created a way to store "meta data", i.e. arbitrary information. The Technology objects don't have this feature currently, like most other objects. In case of the Technology object, I'd suggest you simply keep additional information in a separate dict with the technology name as the key. This should work, as the technology name is a unique identifier for the technology.

Matthias