KLayout/klayout #1980: DRC Python API in Pycharm

EmreCan12345, 2025-01-26:
How to call Klayout in Pycharm to implement DRC of  a layout and return the checking information?This way I can write python code in pycharm to automatically correct the problems in the layout.Thanks!

EmreCan12345, 2025-01-26:
I am a new user of Klayout, please give me some advice

chkkbim, 2025-01-26:
[1] You can read [this](https://github.com/KLayout/klayout/issues/1402) to start with.
I just performed a simple search for "PyCharm" using the search function.

[2] If you need more help, please go to the KLayout [forum.](https://github.com/KLayout/klayout/issues/291)

EmreCan12345, 2025-01-26:
> [1] You can read [this](https://github.com/KLayout/klayout/issues/1402) to start with. I just performed a simple search for "PyCharm" using the search function.
> 
> [2] If you need more help, please go to the KLayout [forum.](https://github.com/KLayout/klayout/issues/291)

Thank you, but I want to know how to do DRC in pycharm

klayoutmatthias, 2025-01-28:
Hi @EmreCan12345,

first of all, this is not a support forum, but an issue tracker.

I understand you want to create a Python function that runs a DRC and somehow delivers the results. There are two options:

1. Don't use DRC scripts. These are Ruby code and naturally don't interface with Python. However, this Ruby code is only a thin convenience layer around a number of classes: `Region`, `Edges`, `EdgePairs` and `Texts`. It is basically possible to rewrite a DRC script as a plain Python script, but at the price of more lines of code. With this approach, you can run the DRC script inside your layout optimization / correction loop.

2. Run DRC in a subprocess - i.e. "klayout <layout-file> -b -r <drc-script>". Let the subprocess generate a report database file or a GDS files. Inside your Python script, read this file into a `ReportDatabase` object or into a `Layout` object and pull the information you need from there.

Matthias

EmreCan12345, 2025-02-01:
> Hi [@EmreCan12345](https://github.com/EmreCan12345),
> 
> first of all, this is not a support forum, but an issue tracker.
> 
> I understand you want to create a Python function that runs a DRC and somehow delivers the results. There are two options:
> 
> 1. Don't use DRC scripts. These are Ruby code and naturally don't interface with Python. However, this Ruby code is only a thin convenience layer around a number of classes: `Region`, `Edges`, `EdgePairs` and `Texts`. It is basically possible to rewrite a DRC script as a plain Python script, but at the price of more lines of code. With this approach, you can run the DRC script inside your layout optimization / correction loop.
> 2. Run DRC in a subprocess - i.e. "klayout  -b -r ". Let the subprocess generate a report database file or a GDS files. Inside your Python script, read this file into a `ReportDatabase` object or into a `Layout` object and pull the information you need from there.
> 
> Matthias

I got it, thank you

closed 2025-02-01 by EmreCan12345